NETMF/netmf-interpreter #293: MetaDataProcessor stack corruption due to COR_ILMETHOD_DECODER incompatibility

elasota, 2015-09-07:
MetaData::Parser::ParseByteCode and MetaData::Parser::Dump_EnumTypeDefs create COR_ILMETHOD_DECODER instances on the stack, and the constructor in turn calls DecoderInit, which is statically linked in from format.lib.

DecoderInit fully initializes the COR_ILMETHOD_DECODER instance, but a "cbLocalVarSig" field was added to it at some point that isn't present in the corhlpr.h in the NETMF Tools dir.
See: https://github.com/dotnet/coreclr/blob/master/src/inc/corhlpr.h

Because of this, stack corruption occurs when MetaDataProcessor is compiled against newer Windows kits and -parse is used.

smaillet-ms, 2015-09-08:
Can you provide a step by step sequence to reproduce the problem?

elasota, 2015-09-08:
Nevermind, turns out it was a bad build that wasn't linking the CorHelper project correctly.

closed 2015-09-08 by elasota