NETMF/netmf-interpreter #294: Support for SDHC above 2 GB?

doingnz, 2015-09-09:
Anyone working on extending support in v4.4 to SD cards above 2 GB capacity?

smaillet-ms, 2015-09-14:
Not that I'm aware of.

cpfister, 2015-09-23:
We have done it for 4.3, in our Mountaineer Prime sources. Speaking hypothetically: if we were to contribute this code, would there be someone who does the port (as far as it may be necessary) to 4.4 and test it, without us having to give tech support (we don't have the time at the moment)?

doingnz, 2015-09-24:
I am happy to have a go adding the changes and testing. 

Originally I contributed the changes to go from SD to SDHC (2GB), so familiar with that code base. 

Are your sources significantly different?

When I last look at it, the remaining changes appeared to be in the file system to complete the process to change lengths and offsets to all be 64 bit values. There is a typedef currently set to a 32 bit signed int that is used most places, but not everywhere. Hence changing it to 64 bit was not enough. Needed a few more places switched to use the correct typedef.

How would you like to share the sources? 
email ZIP: doingnz @ gmail.com
DropBox: https://www.dropbox.com/sh/n8jhrv8obpngn05/AAAVF64oVJnuP_ndlUpjCcD-a?dl=0

Kind regards
Richard