NETMF/netmf-interpreter #306: Running BitConverter tests crashes

mortezag, 2015-09-24:
Running BitConverter tests on MCBSTM32F400 (Keil) crashes and make the board not responsive. I have to erase the deployment region to make it work again.

doingnz, 2015-11-03:
at which test does it crash?

doingnz, 2015-11-03:
The difficulty appears to be the use of reinterpret_cast<T>(v).

In theory the result of the expression reinterpret_cast<T>(v) is the result of converting the expression v to type T. If T is a reference type, the result is an lvalue; otherwise, the result is an rvalue and the l-value to r-value (4.1), array  to pointer (4.2), and function to pointer (4.3) standard conversions are performed on the the expression v. Types shall not be defined in a reinterpret_cast."

In practice reinterpret_cast leads to trouble.

On my iMXS solution, when running the tests, I can get all the way to BitConverterTest_ToBoolean() before the wheels fall off and suddenly Helper.BAToBool(byteArray, 3, true); is trying to compare two boolean values that are both displayed in Visual Studio as "true", and yet the comparison fails. i.e. true != true. 

Under the hood in corlib_native_System_BitConverter.cpp,
``` stack.SetResult_Boolean(*reinterpret_cast<bool*>(p + index));```
can results in undefined values being compared as there are no rules on what to do about memory alignment when converting from 8 bit BYTE to 32 bit boolean as it may be in some systems.   

Attempting to access non aligned 32 bit memory addresses is inviting an exception too.

Can replace
```
	p = pArray->GetFirstElement();
	stack.SetResult_Boolean(*reinterpret_cast<bool*>(p + index));
```
with

```
	p = pArray->GetFirstElement(); 
	b = *(p + index);
	res = (b != 0); // assume any value not zero is true. Zero is false. 
	stack.SetResult_Boolean(res);
```

Define variables before ```FAULT_ON_NULL_ARG(pArray);``` to avoid compiler warnings about uninitialized variables. 

	BYTE *p = NULL,b=0;
	int index = 0;
	bool res = false;

A related question is if void SetBoolean( bool val ) needs to be updated to specifically test val !=0?

```void SetBoolean( bool              val                     ) { CLR_RT_HEAPBLOCK_ASSIGN_INTEGER32_SIGNED( DATATYPE_BOOLEAN, (val!=0) ? 1 : 0 ); }
```

It looks appropriate that we replace all reinterpret_cast<T>(v) with something else in all the methods in ```corlib_native_System_BitConverter.cpp,``` .

smaillet-ms, 2015-11-03:
Thanks for investigating and the detailed report. Your proposed solution sounds like a solid plan. 
reinterpret_cast with pointers and pointer math is trouble for sure as you get into alignment problems if you aren't careful. Unfortunately there isn't a well supported portable means to annotate code as generating unaligned accesses so we can't express the problem directly. 

Go ahead and make a pull request with those fixes.

doingnz, 2016-05-11:
Should this #306 be closed? See #395

Does the #395 update run correctly on MCBSTM32F400?