NETMF/netmf-interpreter #308: 127.0.0.1 does not resolve to local host on MCBSTM32F400

mortezag, 2015-09-25:
Trying to run HttpStatusCodeTests from HttpTests on MCBSTM32F400 port, almost all the tests are failing. All the tests are trying to run a server and then try to send a http request to 127.0.0.1:8080 but the calls fail. When I change the the hard-coded 127.0.0.1 to IPAddress.GetDefaultLocalAddress(), many of tests pass.

lt72, 2015-09-28:
Please check if the loopback driver is included in the image

mortezag, 2015-09-28:
loopback driver is not included and "localhost" does not work either.

smaillet-ms, 2015-09-29:
When you say "localhost" doesn't work either, can you explain more about what you tried? A source example would help to establish a test case.

lt72, 2015-10-06:
http://127.0.0.1/x/y/z does not work, and also http://127.0.0.1
lwIP requires a loopback driver to make the above work
such driver was already part of 4.3, we need to re-establish that under the new model
I think no change at the driver level is necessary