NETMF/netmf-interpreter #316: Cortex-A5 (ATSAMA5D3-XPLD)

csobsidian, 2015-09-28:
Other than being an arduous amount of work, are there technical challenges that exclude .NetMF from running on a Cortex-A5 development board? I have an extra unused Atmel Development board and had some great ideas on what I would like to be able to do with it from a .Net perspective.

cpfister, 2015-09-29:
No, it's indeed just a real lot of work ;-) Most relevant technical difficulty is probably the correct initialization of the memory management unit.

csobsidian, 2015-09-29:
Thank you for your reply! My own assessment was the same. I've read through the entirety of the Cortex-A5 spec and about 90% of the data sheet for the ATSAMA5D36 (speaking of arduous tasks). Although my experience is quite limited, the documentation was very precise with regards to the MMU and I actually expect the driver development to be the more daunting task.

Having only read the spec for the A5, can I assume that my driver code will be "generally portable" to other Cortex-A series systems based on the ARMv7-A spec (i.e. Cortex-A8, -A9, -15)?

lt72, 2015-10-06:
Hi, I fear not.
Here are two examples, for the simplest Cortex-A:
A7: http://infocenter.arm.com/help/topic/com.arm.doc.ddi0464f/BIIJFHJD.html 
A5: http://infocenter.arm.com/help/topic/com.arm.doc.ddi0433a/BIIJFHJD.html 

the MMU is a direct reflection of the memory system, and the further up you go in the processors hierarchy, the more complicated it becomes. 

That said, a lot of features can be skipped, (e.g. security extensions) but still it is a lot of work, and difficult to debug.

csobsidian, 2015-10-06:
Yeah I realized a day or so after I asked that it was an absurd question. I'm thinking that even among various Cortex-A5 implementations most code will likely not be portable. The best portability I will likely get will be between processors in the same class made by the same manufacturer.

cpfister, 2015-10-07:
Yes, the drivers are specific to the microcontroller family.

When we ported NETMF to Cortex-M3 (which, regarding NETMF, is the same as an M4), the most difficult part was the core initialization etc., but the more time-consuming part was to write the drivers for the STM32 peripherals. Once most of them had been done, supporting new STM32 members has become a routine task.

DerekGn, 2015-11-23:
I am currently working on a port to this board. I have a portbooter and tinybooter up and running. I need to test/edbug the flash and USB drivers.