NETMF/netmf-interpreter #326: Base 64 encoding/decoding

josesimoes, 2015-10-13:
The framework Convert class is implemented in a way that is using as default a charset that is not compliant with RFC4648. 
I'm aware that it exposes the property UseRFC4648Encoding that is meant to change the charset.
My questions are:
- why is that the default behaviour is NOT compliant with the related RFC?
- why is that the default behaviour is different from the full .NET framework?
- do we really need this non standard implementation and especially do we need it's default to be the non standard/non compliant option?

I understand that there may be out here code that for some reason is it making use of this but my suggestion would be to make a breaking change here and remove it. Or at most deprecate it and - this is for sure - make the default behaviour the RFC4648 compliant one.

ppatierno, 2015-10-13:
The same here ... I had a problem few months ago using the Base64 encoding for HMAC processing for SAS Token to send to Azure Service Bus. 

http://www.embedded101.com/Blogs/PaoloPatierno/entryid/554/non-standard-base64-encoding-in-the-net-micro-framework

I opened the issue on CodePlex here : http://netmf.codeplex.com/workitem/2385

ppatierno, 2015-10-20:
@smaillet-ms ... I see that you added this issue to the v4.5 Beta version 
Is the 4.4 version so close that we can get it solved ? :-(

smaillet-ms, 2015-10-20:
Yes, we are generating final binaries now and will release shortly. V4.5 will be a shorter time frame with several of the pending PRs coming in right away. With the major focus being on reconciling the CMSIS vs mBed vs current NETMF HAL APIs for the lowest level.