NETMF/netmf-interpreter #327: Incorrect coercion of char to Int32

martincalsyn, 2015-10-13:
consider the code:

string s = "foo";
char c = ' '; // a space
s += c;

Under VS2013, this will result in s == "foo ".  Under VS2015, this will result in s == "foo32".

Stepping through the code, under VS2015, c is being corerced to Int32 and then Int32.ToString() is being called.  Under VS2013, no coercion occurs and char.ToString() is called.

This appears limited to NETMF code generation under VS2015. Code generation for other platforms (Win32, for example) results in the right behavior and code generation under VS2013 is also correct.

martincalsyn, 2015-10-13:
original discussion and code samples : https://www.ghielectronics.com/community/forum/topic?id=21720

smaillet-ms, 2015-10-13:
What happens when targeting the desktop in both cases? For NETMF the VS2015 and VS2013 builds should use the same compiler (e.g. not Roslyn) so a difference in behavior between the two for the same framework version is a bit surprising.

martincalsyn, 2015-10-13:
The IL is different between VS2013 and VS2015. Win32 JIT'er interprets both VS2013 and VS2015 IL correctly (i.e., returns the expected result). The NETMF interpreter interprets VS2013 IL correctly, but returns the wrong result for VS2015 IL.  The IL is here for comparison : https://www.ghielectronics.com/community/forum/topic?id=21720&page=2#msg203841

martincalsyn, 2015-10-13:
Just re-read your comment about VS2015 and 2013 - under the 4.3 SDK they do not appear to be using the same compiler. I have the 4.3 SDK (not 4.4) installed.  If you are forcing the old compiler for 4.4, then yes, this issue would go away for 4.4, but what is the recommendation for 4.3? don't use VS2015 with 4.3?

toxsedyshev, 2015-10-13:
I'm the topic starter from the GHI forum. Subscribed to the issue.

lt72, 2015-10-14:
Seem like we are not supporting yet "constrained"
https://msdn.microsoft.com/en-us/library/system.reflection.emit.opcodes.constrained(v=vs.110).aspx

smaillet-ms, 2015-10-14:
Definitely not, as constrained was added to support generics. Though it is technically orthogonal to generics and might be something we could look into implementing in the future.

martincalsyn, 2015-10-14:
So, it would seem that 4.3 under VS2015 is using the default, newer compiler. Given that there is no fix forthcoming for 'constrained' (and certainly none for 4.3), is there a way to force a different compiler choice when running 4.3 under VS2015, or will there be guidance to not use VS2015 at all with 4.3?

(I am presuming then that this is a non-issue for 4.4 if you are forcing use of the older compiler, but I can't verify that at the moment because I don't have a 4.4 SDK build ready to run)

smaillet-ms, 2015-10-15:
It looks like we didn't find issues with the newer compiler until after launching the 4.3QFE2 SDK. The key component is the MSBUILD support files, which are part of the SDK. Updating that support would effectively require going back to CodePlex and updating the build files and generating a new "QFE3" SDK. As a lower tech alternative someone could figure out the changes needed and then provide a self contained tool to automatically update/path the files in-place. (That would be faster, easier and something the community could do without needing the larger scale new SDK release or MSI patch)

martincalsyn, 2015-10-15:
Ok Steve. Thanks. I'll give it a go to try and create a patch tool or msi. I'll let you know when I have it ready - figure a week or two based on my current workload.

martincalsyn, 2015-10-19:
Until a better fix comes along, I have a powershell script that will fix up existing SDK installations: https://www.ghielectronics.com/community/codeshare/entry/1053