NETMF/netmf-interpreter #331: Any suggestions to resolve why assembly checksum is not changed?

doingnz, 2015-10-20:
I would be very grateful for some ideas on why a checksum for ``spot_io_native.cpp`` in my modified build of MF 4.1 is not changing. In essence I want to use the ``spot_io_native.cpp`` from MF 4.3 in my private build of "MF 4.1".

In MF 4,1 the checksum for ``spot_io_native.cpp`` is 0x21770FF8
In MF 4.3  the checksum for ``spot_io_native.cpp`` is 0xEE0571D9

My solution with the MF 4.3 version of ``spot_io_native.cpp`` (2 changes in this file, and other edits in related files to bring this API to MF 4.3 specification) builds without error. When I deploy with MFDeploy and view the startup debug, I get the following error:

``Invalid native checksum: Microsoft.SPOT.IO 0x21770FF8!=0xEE0571D9``

i.e. when run on the hardware the checksum calculated is still the MF 4.1 value!

Any suggestions highly appreciated as I have been wheel spinning on this for a couple of days trying to figure out what is going on. 

PS: end goal is to make a pull request for the 2GB+ SDHC support (See #294)

Thank you in advance.

doingnz, 2015-10-20:
to confirm its using the correct file, I replaced 0xEE0571D9 ``in spot_io_native.cpp`` with 0xAAAA5555

I then get the expected error:

``Invalid native checksum: Microsoft.SPOT.IO 0x21770FF8!=0xAAAA5555``

This confirms the build is using the updated ``spot_io_native.cpp``, but the Checksum remains 0x21770FF8, when I expect it to be 0xEE0571D9.

smaillet-ms, 2015-10-20:
The checksums are not computed at run time. The check sum in the .PE file is computed at compile time, the one in the CPP file is computed when the stubs are generated and therefore must be manually updated in the source code. (For system libs the normal build will re-generate the stubs but in a separate "stubs" folder in the build output folder. You have to manually merge the updates as there is usually a lot of additional code in the files that it can't be auto-merged. )

doingnz, 2015-10-20:
Blush :-). Thank you for the feedback. 

The Compiler was doing the correct thing and patiently telling me the interface was unchanged. 
In essence I had not merged in the changes under the ``.\Framework\Core\Native_IO`` folders, only the ones in ``.\CLR\Libraries\SPOT_IO''

Thank you ... a few more steps closer to 2GB+ SDHC support

closed 2015-10-20 by doingnz