NETMF/netmf-interpreter #333: Release Build troubles

maiorfi, 2015-10-22:
Hi.

I can't build any firmware solution neither sdk since I get always following errors I can't spot what could depend on.

I uploaded full log at https://dl.dropboxusercontent.com/u/7414592/TinyCLR.it/msbuild_log.zip but it seems that root problem is (please forgive italian msbuild log):

error CS0246: Il nome di tipo o di spazio dei nomi 'CODESIGN' non è stato trovato. Probabilmente manca una direttiva using o un riferimento all'assembly. [D:\NETMF\netmf-interpreter\Framework\Tools\BuildTasksInternal\BuildSigner\BuildSignerSpotBuild.csproj]

i.e. cannot find "CODESIGN.Submitter" assembly...

...any hint?

Thanks!

maiorfi, 2015-10-22:
Ok, it seems that netmf-interpreter\Framework\Tools\BuildTasksInternal\BuildSigner\BuildSignerSpotBuild.csproj contains a bad hint for CODESIGN.Submitter.dll, so changing:

$(MSBuildProgramFiles32)\Microsoft Internal\Codesign.Submitter\CODESIGN.Submitter.dll

with

$(SPOROOT)\tools\x86\CODESIGN\CODESIGN.Submitter.dll

makes msbuild happy again :smile:

closed 2015-10-22 by maiorfi

smaillet-ms, 2015-10-22:
Thanks for the info. We really need to disable or remove that stuff from the non-internal build as it's not something that could ever work outside of our internal infrastructure.

PR #339 (2015-11-12): Fixed build issue for exxternal builds from repository
Fixes #333 
Added additional hint path for code sign submitter so that external builds can succeed (even though signing isn't performed externally the component is needed to complete building the msbuild task dll)