NETMF/netmf-interpreter #334: Converting Long to String

weixiongmei, 2015-10-22:
When converting the Long type variable or value to a string on .Net MF 4.4, it always returns the content "4294967296" as the result, builded with MDK 5.15, VS2015, running on STM32F030CC(Cortex-M0).

cw2, 2015-10-22:
I assume this is your highly customized firmware build - are you sure you have proper support for 64-bit integers? It is not easy to troubleshoot such issues without seeing the actual code, but it could be due to unsupported 64-bit integer printf() format string specifier ("%lld"); or something completely different...

weixiongmei, 2015-10-22:
Thankyou cw2, it's having 64-bit interger problem.... Do you know how to fix it? is it the code problem or build tool setting problem?

cw2, 2015-10-22:
Hm, after giving it a little bit more thought it seems the problem is most likely somewhere else. MDK does not really have any specific settings for 64-bit support (long-long printf formatting) and it should simply work - unless there is something special about its libraries for Cortex-M0, which I could not find anywhere in the documentation. If it was GCC, then I would know what to do, but for MDK I don't know - are you able to place a breakpoint in the code and step through in the debugger? There could be for example stack corruption (too small?)...

As I said previously, it is very hard to troubleshoot such issues without the possibility to reproduce it. I am not forcing you to publish the code, that is your decision and I respect it, but please understand my options are rather limited - you've changed thousands of lines and "something does not work"...

weixiongmei, 2015-10-22:
Hi cw2,
i'll happy to share the codes with you if i still can't get it works with MDK, what you just said i think it's correct, it's releated to the cortex-m0 libraries, i just builded the 4.4 for cortex-m3, the M3 has no problem at all.

And the stack does not corrupt, there's "plenty" of ram for running an simple C# app + firmware. 

Compile the firmware with GCC, the size of the firmware would be much bigger, but i would get a try on it, and see what happen. Please advise what to do with the GCC, config the newlib-nano? Thank you!

Best Regards,
Wilson Mei

weixiongmei, 2015-10-23:
The problem has been fixed with the MDK, everything for the cortex-m0 is working well now, thanks for your advice mr.cw2. It can runs on all the chips of the STM32F0 series now. Ha Ha :)

Regards,
Wilson Mei

cw2, 2015-10-23:
The STM32F4DISCOVERY solution (in NETMF 4.4) uses GCC with newlib-nano, but you'd need GCC with 64-bit (long long) support (which is not enabled by default). I have such toolchain available for download in my repo, please see the links in issue #164.

I have not yet had a chance to make the most recent GCC release, it will come in a few days or so...

closed 2015-11-29 by smaillet-ms