NETMF/netmf-interpreter #335: How to build?

roceh, 2015-10-22:
Having some problems trying to build 4.4 - I assume it is building with vs2015 and launchpad GCC? When I attempt to build it seems to require some additional tools and errors out early looking for some sort of codesign app - are these available? 

I've seen mention of build_tools.zip in some other issues, is this still needed, if so is there a link? Apologies if this is in the wiki somewhere - I did have a search to see if there was any info.

smaillet-ms, 2015-10-22:
Please see #333 - hopefully that address your issues. Though you shouldn't need to build the SDK as we have a prebuilt and signed binary available in the VS gallery and also from the GitHub [releases](https://github.com/NETMF/netmf-interpreter/releases) page

roceh, 2015-10-22:
oops didn't notice an issue had already been opened. thanks for the info.

smaillet-ms, 2015-10-22:
No problem, we had to make some late changes for the code signing stuff and missed verifying it could still build on a system without the full access and support infrastructure for signing used internally. I'll try and get a fix for that checked in within a few days.

roceh, 2015-10-22:
Got a bit further - found the build_tools.zip via googling around ( for others its here http://netmf.github.io/downloads/build-tools.zip) I think it used to be mentioned on the readme in a previous commit.

Placed the tools in the zip in the tools folder above the netmf-interpreter folder which has the netmf source.

Got the crypto msi installed ( https://github.com/NETMF/netmf-interpreter/releases/download/NETMFCryptoLibraries/NetMFCryptoLibs.msi )  and put the libs into the crypto\libs folder in the netmf source folder. 

build_sdk now fully builds.

Tried to build the STM32F4DISCOVERY solution using launchpad GCC 4.9.3 and its gets quite far, but fails when it tries to link the crypto libs - I assume looking at the crypto folder libs they are only for the keil compilers? Is there anyway to build tinybooter without the crypto so i can use GCC?

smaillet-ms, 2015-10-23:
The requirement for the bin tools is documented in the [WIKI getting started topic ](https://github.com/NETMF/netmf-interpreter/wiki/Getting%20Started) step number 4. Though if for some reason that wasn't clear or we can do something to help make that more discoverable we'd love to hear it. 

The libs are ARM ABI compliant so should work with either compiler.

roceh, 2015-10-23:
Ah ... I somehow missed that doc =)

I've got tinyclr/tinybooter compiled now, although it does fail on the signing. 

```
BuildSigFiles:
  Create Signature files for C:\netmf\netmf-interpreter\BuildOutput\THUMB2FP\GCC4.9\le\FLASH\release\STM32F4DISCOVERY\bin\Tinybooter.bin
  C:\netmf\netmf-interpreter\BuildOutput\public\Release\Server\dll\MetaDataProcessor.exe -sign_file C:\netmf\netmf-interpreter\BuildOutput\THUMB2FP\GCC4.9\le\FLASH\release\STM32F4DISCOVERY\bin\Tinybooter.bin C:\netmf\netmf-interpreter\tools\bin\tinybooter_private_key.bin C:\netmf\netmf-interpreter\BuildOutput\THUMB2FP\GCC4.9\le\FLASH\release\STM32F4DISCOVERY\bin\Tinybooter.sig
MMP : error MMP0000: CLR_E_FAIL [C:\netmf\netmf-interpreter\Solutions\STM32F4DISCOVERY\TinyBooter\TinyBooter.proj]
C:\netmf\netmf-interpreter\tools\targets\Microsoft.SPOT.System.GCC.targets(348,5): error MSB3073: The command "C:\netmf\netmf-interpreter\BuildOutput\public\Release\Server\dll\MetaDataProcessor.exe -sign_file C:\netmf\netmf-interpreter\BuildOutput\THUMB2FP\GCC4.9\le\FLASH\release\STM32F4DISCOVERY\bin\Tinybooter.bin C:\netmf\netmf-interpreter\tools\bin\tinybooter_private_key.bin C:\netmf\netmf-interpreter\BuildOutput\THUMB2FP\GCC4.9\le\FLASH\release\STM32F4DISCOVERY\bin\Tinybooter.sig" exited with code 10. [C:\netmf\netmf-interpreter\Solutions\STM32F4DISCOVERY\TinyBooter\TinyBooter.proj]
```

Also I did have to change the Microsoft.Spot.system.gcc.targets FLOATING_POINT_FLAG from "-mfloat-abi=hard -mfpu=fpv4-sp-d16" to "-mfloat-abi=softfp -mfpu=fpv4-sp-d16" and the GCC_ARCH_DIR to softfp only otherwise I got these errors:

```
c:/gccarm/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe : error : C:\netmf\netmf-interpreter\BuildOutput\THUMB2FP\GCC4.9\le\FLASH\release\STM32F4DISCOVERY\bin\Tinybooter.axf uses VFP register arguments, Crypto.lib(rsa_zenith.obj) does not [C:\netmf\netmf-interpreter\Solutions\STM32F4DISCOVERY\TinyBooter\TinyBooter.proj]
  c:/gccarm/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file Crypto.lib(rsa_zenith.obj)
c
```

cgopalg, 2015-11-30:
I have got tinyclr/tinybooter compiled, but it does fail on the signing.
Any solution for the error "MMP : error MMP0000: CLR_E_FAIL".

Using "Exec" task from assembly "Microsoft.Build.Tasks.Core, Version=14.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a".
Task "Exec"
  H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\BuildOutput\public\Release\Server\dll\MetaDataProcessor.exe -sign_file H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\BuildOutput\THUMB2FP\MDK5.06\le\FLASH\release\MCBSTM32F400\bin\Tinybooter.bin H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\tools\bin\tinybooter_private_key.bin H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\BuildOutput\THUMB2FP\MDK5.06\le\FLASH\release\MCBSTM32F400\bin\Tinybooter.sig
MMP : error MMP0000: CLR_E_FAIL [H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\Solutions\MCBSTM32F400\TinyBooter\TinyBooter.proj]
H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\tools\targets\Microsoft.SPOT.System.MDK.targets(386,5): error MSB3073: The command "H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\BuildOutput\public\Release\Server\dll\MetaDataProcessor.exe -sign_file H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\BuildOutput\THUMB2FP\MDK5.06\le\FLASH\release\MCBSTM32F400\bin\Tinybooter.bin H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\tools\bin\tinybooter_private_key.bin H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\BuildOutput\THUMB2FP\MDK5.06\le\FLASH\release\MCBSTM32F400\bin\Tinybooter.sig" exited with code 10. [H:\github\netmf-interpreter-4.4-RTW-20-Oct-2015\Solutions\MCBSTM32F400\TinyBooter\TinyBooter.proj]
Done executing task "Exec" -- FAILED.
Done building target "BuildSigFiles" in project "TinyBooter.proj" -- FAILED.
Done executing task "CallTarget" -- FAILED.
Done building target "CoreSystemBuild" in project "TinyBooter.proj" -- FAILED.
Done executing task "CallTarget" -- FAILED.
Done building target "Build" in project "TinyBooter.proj" -- FAILED.
Done Building Project