NETMF/netmf-interpreter #337: How can I check if there are memory leak?

techcap, 2015-10-23:
I suspect there are some memory leak in my project.
It operates during about two days and stops.
I'm testing by removing components one by one. But it takes too long.
Is there any tip for checking if there are memory leak or not?
For example, stack pointer, heap pointer or other registers?

techcap, 2015-10-25:
And can I check GDI object count?

smaillet-ms, 2015-10-26:
Please note the is the .NET Micro Framework, which is an interpreted runtime for .NET applications on small footprint micro controllers. Your question, in particular mentioning GDI objects, suggests you are working with a different runtime as the .NET MF doesn't have GDI objects as there, generally speaking, isn't an OS underneath.

techcap, 2015-10-27:
I ported NETMF 4.4 to my product which have SDRAM and touch LCD.
The problem is that it stops after 1~2 days. 
It looks because of memory leak problem. 
On the managed side, I checked the memory usage using Debug.GC(false), and there is no leak.
When the system stopped, I checked in MDK and there looked native memory assignment problem. 
So now I'm guessing the memory leak in native side.
On latest test, I changed DeviceCode\PAL\Graphics\dotNetMF.proj to stubs, DeviceCode\PAL\Graphics\stubs\dotNetMF.proj. After changed, netmf doesn't stop during 3 days. The test is on going.
So I guess the problem is related to Graphics. Testing is so inefficient and boring because it takes too long time to check if the system will be stopped or not. So I want to know if there are any tips to check memory leak especially in the native side.
Thanks.

smaillet-ms, 2015-11-19:
There isn't any built-in tool for tracking leaks in native code. You could however add a completion in the system to periodically check the heap for available memory and print the results out the debug port so you can monitor it.

techcap, 2015-11-20:
Can you give me some tips for getting available heap or stack memory? I searched on google, but only found third party tool information.