NETMF/netmf-interpreter #338: How to reduce or remove the string table?

weixiongmei, 2015-10-25:
There's a StringTableData.cpp, it used a lot of space of flash, is it possible to remove or reduce the size of it? What's it for? Thank you~

cw2, 2015-10-26:
Yes, it is possible - please have a look at issue #56 and the related pull request #284, its comment contains instructions how to rebuild the string table.

weixiongmei, 2015-10-26:
hi cw2,

i got an error when running this command
msbuild Tools\Targets\Microsoft.SPOT.Support.Settings /t:buildstring

it shows '-loadstrings' is not an internal or external command, and it's not an excutable application.  Any idea? Thanks a lot~~~

cw2, 2015-10-26:
You have to set PRG_MMP environment variable to metadataprocessor.exe path
```
set PRG_MMP=X:\NETMF\BuildOutput\ ??? \metadataprocessor.exe
```

weixiongmei, 2015-10-27:
Thank you, it works~~~