NETMF/netmf-interpreter #348: NETMF Debugger USB driver issue with VS2015

maiorfi, 2015-11-17:
We're currently working with two different STM32F4-based boards, with slightly different porting code, both using .NETMF 4.3.

On the former, we can connect both with VS2013 and with VS2015 (and both with MFDeploy related to 4.3 SDK and with 4.4 one), while on the latter we can only connect with VS2013 and MFDeploy 4.3, since "target board" dropdown is empty if USB transport is selected.

On development machine I have both VS2013, VS2015, SDK 4.3 and SDK 4.4, with respective VSIX installed as well.

I suspect that thi could be related to Legacy WinUSB option in some way, but every time I check that VS and MFDeploy crash (the latter reporting an "exception while enumerating...").

How does SDK (i.e. MFDeploy and VS Extension) recognize USB device suitable for debug/deploy exactly?

Thanks!

smaillet-ms, 2015-11-17:
The legacy WinUSB option was a backwards compat thing for a few devices that used WinUSB but did not report the correct interface GUID. In older versions of the tools MFDeploy and VS would enumerate all WinUSB devices and, if they had a string descriptor at a well known index (5 if I recall correctly without looking at the code) then it was considered a NETMF capable device. Obviously this is a bit overly presumptive and you could regularly find other devices that had a string at that index which would show up and we'd try to use. The results of which were either a benign fail or something going horribly wrong. Either way that's a bad experience. In V4.4 we changed the default enumeration to only show those devices which properly publish the NETMF debugger driver interface GUID. This GUID can reside in the USB configuration in the device (like MCBSTM32F400) which results in a seamless no driver install experience. (e.g. no custom driver needed - not even an INF). Some devices may not have that in their images correctly (as previous NETMF releases had bugs in the reference implementation for that) so for those a custom (and properly signed) INF is still required. That INF should publish the NETMFDBG interface GUID for the device so that it can be found. 

If you own the device firmware implementation on these devices then I'd recommend updating the USB configuration data structures to match the reference (changing VID/PID and product names as appropriate, of course) so you can have a true plug and play experience without the hassles of driver signing.

maiorfi, 2015-11-18:
Hi Steve.

I took a look at different Solutions\SolutionXXX\DeviceCode\usb_config.cpp I have for 4.3, and two reference porting solutions for 4.4, and I didn't spot any significant difference, with the notable exception that the only board I can connect to both with 4.3 and 4,4 MFDeploy tool versions (and both with VS2013 and VS2015) uses USB 1.1 descriptor instead of 2.0. Could it explain my issue?
Also, these tests showed that neither DIscovery4 nor Keil board with 4.4 porting (latest code state on dev branch) can be "seen" by (my) MF Deploy 4.4 and VS2015, but only by MFDeploy 4.3 and VS2013.

Any hint?

smaillet-ms, 2015-11-18:
Keil MCBSTM32F400 as built from GitHub sources should be visible in MFDeploy and VS without enabling the legacy WinUSB support. That's the primary platform we built and tested everything on/with. What OS version are you running? Does the device even show up in Device manager? (e.g. are you chasing a cable or port problem and not a USB descriptor problem? )

I just ran a diff between v4.3 MCBSTM32E and the v4.4 MCBSTM32F400 usb_config.cpp and there are definitely differences, the two key differences/bug fixes from previous releases are:
 1. The Device Descriptor has version 2.0, which is required for the extended descriptor support to work
 2. The extended os property descriptor flags (line 255 of the MCBSTM32F400 version) has an additional flag (USB_REQUEST_TYPE_INTERFACE) which is required for the USB driver support to retrieve the extended descriptor correctly.

maiorfi, 2015-11-20:
Hi.

I have rebuilt my porting solution adding USB_REQUEST_TYPE_INTERFACE flag (usb version was already 2.0), but VS2015/MFDeploy4.4 cannot detect USB driver, that is correctly recognized by Windows as WinUSB driver.

For sake of completeness, I also tried to switch to USB1.1 configuration, but as stated in usb_conf.h source file, WinUSB driver cannot be used in that case, and I didn't suceed in loading mfwinusb legacy driver in that case (probably I should cook a specific .inf file).

Any suggestion?

maiorfi, 2015-11-24:
I finally spotted the cause of my issue.

The short story is that issue depends on proper driver.

Long story is that using an .inf file not referring to a .cat file causes Windows install USB peripheral as USB Device, while using a .inf+.cat driver let Windows install USB debug interface under "USB Controllers". I do not actually know whether such different classification among peripherals is the real cause, but same device-side USB configuration enables board detection in VS2015 once I install Mountaineer (Ethernet) official drivers (of course I had to change USB PID code inside usb_config.cpp, but board I'm working on is based on same porting source base, after all :smile: ).

closed 2015-11-24 by maiorfi

smaillet-ms, 2015-11-24:
With v4.4 you should not need to use an INF or .CAT file, that's ding things the hard way (and more expensive way as you'll need to get a digital signature, and WHQL test certification, etc...) The whole point of using the extended OS descriptors is to avoid all of that for anything that can be WinUSB or HID supported.