NETMF/netmf-interpreter #349: STM32F411 AnalogInput Problem

techcap, 2015-11-18:
I tested cw2's stm32f411nucleo branch on NUCLEO-F411RE board.
Everything may look good. 
But when I used AnalogPort.Read(), it hangs in L:110 in STM32F4_AD_functions.cpp.
WIll it be related to core clock?
This branch used core clock to 96MHz not 100MHz.

MF
----------------------------------------------------------------------------------------------------------------------
                AnalogInput input = new AnalogInput(Cpu.AnalogChannel.ANALOG_0);

                Debug.Print(input.Read().ToString());

FW : STM32F4_AD_Functions.cpp
-------------------------------------------------------------------------------------------------------------------------

INT32 AD_Read( ANALOG_CHANNEL channel )
{
    int chNum = g_STM32F4_AD_Channel[channel];
  
    // check if this channel is listed in the STM32F4_AD_CHANNELS array
    for (int i = 0; i < STM32F4_AD_NUM ; i++) {
        if (g_STM32F4_AD_Channel[i] == chNum ) {
            // valid channel
            int x = ADCx->DR; // clear EOC flag

            ADCx->SQR3 = chNum; // select channel
        
            // need to enable internal reference at ADC->CCR register to work with internally connected channels 
            if(chNum == 16 || chNum == 17) {
                ADC->CCR |= ADC_CCR_TSVREFE; // Enable internal reference to work with temperature sensor and VREFINT channels
            }
    
            ADCx->CR2 |= ADC_CR2_SWSTART; // start AD
///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
            while (!(ADCx->SR & ADC_SR_EOC)); // wait for completion <== L:110 it hangs here!!!
///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
            // disable internally reference
            if(chNum == 16 || chNum == 17) {
                ADC->CCR &= ~ADC_CCR_TSVREFE; 
            }
    
            return ADCx->DR; // read result
        }
    }

    // channel not available
    return 0;
}

techcap, 2015-11-18:
NUCLEO-F411RE doesn't have external crystal. 
But the strange thing is STM32F4_BootstrapCode() in STM32F4_bootstrap.cpp stops HSI.
And STM32F4_bootstrap.cpp enforces SYSTEM_CLOCK_HZ should be divided 48000000.

cw2, 2015-11-18:
ad 96 MHz) The USB peripheral requires 48 MHz clock and there are certain limitations on how it can be achieved, based on the core clock frequency and allowed PLL prescaler value combinations. I don't remember all the details for this particular microcontroller variant, but I think 48 MHz clock cannot be divided from 100 MHz with enough precision for the USB peripheral to operate reliably, so the core has to run at 96 MHz. The maximum frequency can be set only if you don't want to use the USB peripheral (which is fine in NETMF, just have serial transport for deployment (and hope for good prescalers for baudrates : )).

ad Nucleo external clock) While the Nucleo board does not have an external crystal oscillator, it uses [8 MHz] external clock from the onboard ST-LINK/V2 debugger. This has the same clock source configuration as for the external crystal, so there are no changes required in the NETMF code. (Technically, it is recommended to set the HSE bypass bit, but it works without it).

ad analog conversion hang) I don't have any idea why the conversion does not end, but most likely there is something wrong with the ADC configuration. Please note my ports have been tested only for the most basic operations, which means application deployment from Visual Studio and basic GPIO (LED blinking). This is also the reason why they don't have much publicity and were not included in the official code base. I would probably look at AD_Initialize and check the settings there...

techcap, 2015-11-19:
Thanks for your kind and detail explanation. It was really helpful for understanding.

I found the problem. Nucleo board has ADC1 not 3. So I modified STM32F4_ADC to 1 and it worked properly.

And Nucleo board doesn't have DA, so I changed it to stub.

Thank you very much.

Do you know any reason why MF doesn't update Firmware Package for Family STM32F4 which is supplied from ST? In the case of stm32f4xx.h, the one in MF has version v1.2.1. But the latest version is v2.4.1.

closed 2015-11-25 by techcap