NETMF/netmf-interpreter #350: Tools for testing implementation?

DerekGn, 2015-11-18:
Are there any tools available for testing a custom implementations?

I am specifically looking for something to help test a flash driver implementation. I would like to be able to perform write/read/validate of a flash block so that I can debug the driver.

Thanks

DerekGn, 2015-11-23:
I am thinking it should be possible to use the MFDeploy API, specifically the MFConfigHelper class to read and write a configuration item to and from the flash

uscomrs, 2015-11-23:
Rough guide to adding new menu item to MFDeploy to read/write a costume config item. These are not necessarily 100% all the steps. Intended to give you a pointer in the right direction to add a menu item to MFDeploy to read / write a custom config block. This will only read/write into the flash config flash region of the target device.

Look at MFUsbConfig.cs in MFDeployEngine project. You can copy and rename this class to something of your own choice (e.g. MFTestConfig). It stores a single string in a config block. Change the config name string c_name = "TestConfig";

Copy MFConfigUsbDialog.cs in MFDeploy project. (e.g. MFConfigTestDialog). Change to refer to your class above (MFTestConfig.)

Add Menu item in MFDeploy (updateTestConfigToolStripMenuItem) and set its on click even to point to 
```private void OnMenuItem_Click(System.Object sender, System.EventArgs e)```

Add ```
if (mi == updateTestConfigToolStripMenuItem)
                {
                    dlg = new MFConfigTestDialog(device);
                }
```

You can search into the firmware for the corresponding code to read the named USB config in the embedded firmware. It would be nice if this was exposed up into the CS on the firmware side, but only native code at the moment.

Hope this helps

DerekGn, 2015-11-24:
Thanks,

I was looking for something in the code base/sdk to use rather than write.

I used something like the following to read/write/verify

```C#
using System;
using System.Collections.Generic;
using System.Linq;
using Microsoft.NetMicroFramework.Tools.MFDeployTool.Engine;

namespace TestFlashDriver
{
    class Program
    {
        static void Main(string[] args)
        {
            var config = new List<Byte> {0x55, 0xAA, 0x55, 0xAA};

            try
            {
                using (var deploy = new MFDeploy())
                {
                    using (MFDevice device = deploy.Connect(new MFSerialPort("COM", "1")))
                    {
                        if (device.ConnectToTinyBooter())
                        {
                            using (var configHelper = new MFConfigHelper(device))
                            {
                                do
                                {
                                    configHelper.WriteConfig("Test", config.ToArray());

                                    var result = configHelper.FindConfig("Test");
                                    
                                    if (config.SequenceEqual(result))
                                    {
                                        Console.WriteLine("\nConfig data did not match");

                                        foreach (var b in result)
                                        {
                                            Console.WriteLine(" {0} ", b);
                                        }
                                    }

                                    config.Reverse();

                                } while (CheckForQuit());    
                            }
                        }
                    }
                }
            }
            catch (Exception exception)
            {
                Console.WriteLine("Unhandled Exception: {0}", exception);
            }
        }

        private static Boolean CheckForQuit()
        {
            Boolean quit = false;

            if (Console.KeyAvailable)
            {
                ConsoleKeyInfo consoleKeyInfo = Console.ReadKey();

                if (consoleKeyInfo.Key == ConsoleKey.Q)
                {
                    quit = true;
                }
            }

            return quit;
        }
    }
}
```

closed 2016-01-09 by DerekGn