NETMF/netmf-interpreter #353: Bin2Srec.exe build

DerekGn, 2015-11-27:
The previous version of .net microframework porting kit 4.3 included bin2srec.exe in binary form.
The current master branch has bin2srec source code but it is not built.

smaillet-ms, 2015-11-27:
Do you get some sort of error when you try to build it? Everything that was in the porting kit is now source only. So nothing is binary in the repository

DerekGn, 2015-11-27:
Yes the path to the signing key is incorrect or else I am building it with incorrect build environment vars etc.

Error	2	Cryptographic failure while signing assembly 'd:\Git\NetMf\netmf-interpreter\tools\binToSrec\obj\Debug\binToSrec.exe' -- 'Error reading key file 'd:\framework\key.snk' -- The system cannot find the path specified. '	D:\Git\NetMf\netmf-interpreter\tools\binToSrec\CSC	binToSrec

I was looking through the build scripts to see if it is to be built as part of the MF build, like buildhelper.exe for example. I have not found it in the scripts yet. To be honest this low priority for me as I worked around the problem,and continued on with my porting, just thought I would ask to see what the correct way to build it is.

smaillet-ms, 2015-11-27:
Hmm, I think the "correct way" would be to submit a pull request that changes the project to not use a strong name key :grin: 
Since the porting kit isn't generated anymore any tools provided in source don't need a strong name key. (Otherwise we'd have to resort to, building, signing and distributing them, which seems a waste of effort in comparison to just turning off the strong name signing)

DerekGn, 2015-11-28:
Ok thanks, when I searched I did find the file framework\key.snk thats why I assumed the my build env was wrong.

Just out of curiosity whats the Framework\key.snk used to sign if at all?

smaillet-ms, 2015-11-29:
That key shouldn't be used for anything anymore. (Given that it was published in CodePlex with the public and private parts intact it was pretty much worse than useless). Since we no longer build or ship the PK there's no need to sign any of those tools any more.

closed 2015-11-29 by smaillet-ms