NETMF/netmf-interpreter #358: Explanation of system and slow clock configuration

DerekGn, 2015-12-07:
I am looking for an explanation of clock configuration in the platform_selector.h file. I am hoping that someone can help me out as I have not found too much information on what the various elements are used for.

I have the following configuration

`#define SYSTEM_CLOCK_HZ                 13200000`
`#define SYSTEM_CYCLE_CLOCK_HZ           SYSTEM_CLOCK_HZ`
`#define CLOCK_COMMON_FACTOR             1000000`
`#define SLOW_CLOCKS_PER_SECOND          32768`
`#define SLOW_CLOCKS_TEN_MHZ_GCD         2000000`
`#define SLOW_CLOCKS_MILLISECOND_GCD     1000`

I used values generated by the solution wizard as a base, I have observed during testing of my port is that the following function, which seems common to all implementations that I have seen, overflows.

`UINT64 CPU_TicksToTime( UINT64 Ticks )`
`{`
`    Ticks *= (TEN_MHZ               /SLOW_CLOCKS_TEN_MHZ_GCD);`
`    Ticks /= (SLOW_CLOCKS_PER_SECOND/SLOW_CLOCKS_TEN_MHZ_GCD);`

`    return Ticks;`
`}`

When I set the SLOW_CLOCKS_PER_SECOND to SYSTEM_CLOCK_HZ the operation calculates the correct values in 100ns base time increments.

For example the period of the SYSTEM_CLOCK_HZ at 132 MHz is ~ 7.5ns. When CPU_TicksToTime is called with a value of 14 the result is 1.

Any guidance in how to configure the clocks for a SYSTEM_CLOCK_HZ and SLOW_CLOCKS_PER_SECOND appreciated.

DerekGn, 2015-12-08:
Looks like I used a clock that was off by a factor of 10

cw2, 2015-12-08:
As you already know, `SYSTEM_CLOCK_HZ` determines the core CPU frequency. `SYSTEM_CYCLE_CLOCK_HZ` determines the AHB frequency, it is used to calculate the value of HPRE prescaler (see STM32[F4]_bootstrap.cpp).

`SLOW_CLOCKS_PER_SECOND` is the frequency of the system timer. In a 132 MHz system you'd probably want it to be something like 1 MHz, which gives you timer resolution/granularity of 1 µs.

`CLOCK_COMMON_FACTOR`, `SLOW_CLOCKS_TEN_MHZ_GCD` and `SLOW_CLOCKS_MILLISECOND_GCD` are [Greatest Common Divisor](https://en.wikipedia.org/wiki/Greatest_common_divisor) values, as follows:
```
CLOCK_COMMON_FACTOR         = GCD(SYSTEM_CLOCK_HZ, 1000000)         // 1M
SLOW_CLOCKS_TEN_MHZ_GCD     = GCD(SLOW_CLOCKS_PER_SECOND, 10000000) // 10M
SLOW_CLOCKS_MILLISECOND_GCD = GCD(SLOW_CLOCKS_PER_SECOND, 1000)     // 1k
```
These are needed to ensure the various calculations are whole integers (without a remainder).

So, for your system I would use
```
#define SYSTEM_CLOCK_HZ             132000000 // 132 MHz
#define SYSTEM_CYCLE_CLOCK_HZ       SYSTEM_CLOCK_HZ
#define CLOCK_COMMON_FACTOR           1000000 // GCD(SYSTEM_CLOCK_HZ, 1M)
#define SLOW_CLOCKS_PER_SECOND        1000000 // 1 MHz
#define SLOW_CLOCKS_TEN_MHZ_GCD       1000000 // GCD(SLOW_CLOCKS_PER_SECOND, 10M)
#define SLOW_CLOCKS_MILLISECOND_GCD      1000 // GCD(SLOW_CLOCKS_PER_SECOND, 1k)
```
With the above values `CPU_TicksToTime(14)` returns 140 (the time is represented as the  the number of 100 nanosecond intervals, and one 1 µs tick is ten 100-ns intervals).

cw2, 2015-12-08:
Note: `CPU_TicksToTime` and similar functions are based on the system timer, not the core CPU frequency - probably a little bit confusing names...

DerekGn, 2015-12-09:
Thanks for the info CW very helpful, it clears up my misunderstanding.

I had incorrectly assumed that the SYSTEM_CLOCK_HZ was the system/bus/peripheral clock not the CPU clock and the slow clock was the RTC 32k tick.

One additional question, I am porting to a CPU that can run the timer at a high clock speed 66mhz, would it be useful to have a faster clock speed for higher resolution?

cw2, 2015-12-09:
Regarding the system timer at 66 MHz, I don't think that is needed, unless of course you have a specific scenario that would require it, like obtaining very precise timestamps...

DerekGn, 2015-12-09:
I will divide the clock speed down to a couple of mhz as i don't really have a use case for such a high clock speed.

closed 2015-12-09 by DerekGn