NETMF/netmf-interpreter #362: Fail to build build.dirproj

kmshim, 2015-12-24:
@smaillet-ms 
1. msbuild /t:rebuild /p:flavor=release build.dirproj
2. Error message
cscript //Nologo C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\tools\scripts\Preprocess.js /TemplateFile:C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\tools\AssemblyInfo.cs.template /Properties:"__ASSEMBLYTITLE__=#Microsoft.SPOT.Debugger#;__ASSEMBLYCOMPANY__=#Microsoft Corporation#;__ASSEMBLYCOPYRIGHT__=#Copyright (C) Microsoft Corporation.  All rights reserved.#;__ASSEMBLYDESCRIPTION__=##;__ASSEMBLYVERSION__=#4.4.0.0#;__PRODUCTVERSION__=#4.4.0.0#;__PRODUCTVERSION_CSV__=4, 4, 0, 0;__ASSEMBLYFILEVERSION__=#4.4.0.0#;__FILEVERSION_CSV__=4, 4, 0, 0;__ASSEMBLYCONFIGURATION__=#release#;__ASSEMBLYPRODUCT__=#Microsoft .NET Micro Framework#;__ASSEMBLYTRADEMARK__=#.NET is a trademark of Microsoft Corporation.#;__ASSEMBLYINTERNALNAME__=#Microsoft.SPOT.Debugger#" /OutputFile:bin\Release\Microsoft.SPOT.Debugger_assemblyinfo.txt
C : \Users\shimkyoman\Documents\GitHub\netmfInterpreter\tools\scripts\Preprocess.js(63, 9) Microsoft JScript runtime error : Path not found [C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\Framework\Debugger\Debugger.csproj]

3. this error is cause <OutputPath>bin\Release\</OutputPath> of Debugger.csproj. So I propose that output path should be removed or commented.

smaillet-ms, 2015-12-31:
Can you provide more detailed steps to reproduce this issue? I'm uncleare on how you set up your environement (e.g. which setenv_xxx was used, what directory was the msbuild command executed, etc...) [It's safe to assume a step 0 where the source code is cloned from Github, but nothing else should be assumed]

kmshim, 2015-12-31:
#### OS
> Windows 7 32bit

#### Tool chain
1. GCC
> C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter>setenv_gcc.cmd 5.2.1 "c:\P
rogram Files\GNU Tools ARM Embedded\5.2 2015q4"
> Compiler: GCC 5.2.1 c:\Program Files\GNU Tools ARM Embedded\5.2 2015q4
setting vars for GCC compiler GCC5.2.1

2. Visual Studio
> Visual Studio 2015 Community
> Microsoft (R) Build Engine 버전 14.0.24723.2]
> Copyright (C) Microsoft Corporation. All rights reserved

#### Cloned source information
> Revision: eb4596e9a599c8b62498bca22b78caa825202e0e
> Author: Steve Maillet <smaillet-ms@users.noreply.github.com>
> Date: 2015-11-30 오전 2:12:52
> Message:
> Merge pull request #354 from DerekGn/dev
> Removed assembly signing key
> Modified: tools/binToSrec/binToSrec.csproj
[image]

smaillet-ms, 2015-12-31:
GCC 5.2.1? That's likely going to be a problem either way. We've only built or tested with 4.9x variants.

kmshim, 2015-12-31:
Thanks for your rapidly reply.
But I think that the problem is regardless with GCC version.
I mean that intermediate output files such as dll, pdb and cache should be located at __COMMON directory. Am I wrong?

smaillet-ms, 2015-12-31:
Sorry, I wasn't saying that was the problem, just that it is likely to become one once this issue is resolved. I don't know what the cause of this issue could be other than possibly the use of a 32 bit OS. We have done no testing or validation of the framework build process on 32 bit systems. So there could well be isues there.

kmshim, 2015-12-31:
I always thanks for your contribution.
Happy new year.