NETMF/netmf-interpreter #363: How to fix when input file of MetaProcessor.exe is zero size.

kmshim, 2015-12-27:
1. Reproduce condition
 -. OS: Windows 7, 32bit
 -. Toolchain: launch pad GCC 5.2.1
 -. build command: msbuild /v:diag /p:flavor=release;memory=flash
2. Error message
Command=C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\BuildOutput\public\Release\Server\dll\MetaDataProcessor.exe -sign_file C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\BuildOutput\THUMB2FP\GCC5.2\le\FLASH\release\STM32F4DISCOVERY\bin\tinyclr.bin\ER_DAT C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\tools\bin\tinybooter_private_key.bin C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\BuildOutput\THUMB2FP\GCC5.2\le\FLASH\release\STM32F4DISCOVERY\bin\tinyclr.hex\ER_DAT.sig(작업 ID:2537)
  C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\BuildOutput\public\Release\Server\dll\MetaDataProcessor.exe -sign_file C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\BuildOutput\THUMB2FP\GCC5.2\le\FLASH\release\STM32F4DISCOVERY\bin\tinyclr.bin\ER_DAT C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\tools\bin\tinybooter_private_key.bin C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\BuildOutput\THUMB2FP\GCC5.2\le\FLASH\release\STM32F4DISCOVERY\bin\tinyclr.hex\ER_DAT.sig(작업 ID:2537)
MMP : error MMP0000: CLR_E_FAIL [C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\Solutions\STM32F4DISCOVERY\TinyCLR\TinyCLR.proj]
C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\tools\targets\Microsoft.SPOT.System.GCC.targets(349,5): error MSB3073: "C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\BuildOutput\public\Release\Server\dll\MetaDataProcessor.exe -sign_file C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\BuildOutput\THUMB2FP\GCC5.2\le\FLASH\release\STM32F4DISCOVERY\bin\tinyclr.bin\ER_DAT C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\tools\bin\tinybooter_private_key.bin C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\BuildOutput\THUMB2FP\GCC5.2\le\FLASH\release\STM32F4DISCOVERY\bin\tinyclr.hex\ER_DAT.sig" 명령이 종료되었습니다(코드: 10). [C:\Users\shimkyoman\Documents\GitHub\netmfInterpreter\Solutions\STM32F4DISCOVERY\TinyCLR\TinyCLR.proj]

3. In this case, size of ER_DAT.sig is zero. How can I fix it.

josesimoes, 2016-01-12:
Can be fixed with PR #370

kmshim, 2016-01-13:
I think that this issue may have been solved by @josesimoes. refer to #370.

smaillet-ms, 2016-02-03:
keeping this open until the PR is merged into repo

kmshim, 2016-02-03:
Ok, I got it.

josesimoes, 2016-02-19:
I think that now that PR #370 was merged this one can be closed.

closed 2016-02-19 by smaillet-ms