NETMF/netmf-interpreter #367: Build Date wrong for 2016

SytechDesigns, 2016-01-09:
Last month ( Dec 2015), every port I built, when tested with MFDeploy (Device Capabilities) reported a software version build date that was correct.
However every thing I have built since Jan 1 2016 reports a build date of Dec 31 2015 - irrespective of the build date. Any one have any ideas on this?

SytechDesigns, 2016-01-11:
How does the project constant __DATE__ get set in the build process?

SytechDesigns, 2016-01-11:
That should be __ DATE__

smaillet-ms, 2016-01-11:
__DATE__ is a compiler generated pre-processor macro that is set to the date of the compilation. Thus, if you have libraries built on different days without doing a clean build of them all they will all have different values for __DATE__. Try doing a clean build, and you should see the current date correctly.

closed 2016-02-17 by smaillet-ms