NETMF/netmf-interpreter #369: MCBSTM400 Ethernet problem

SytechDesigns, 2016-01-11:
I have built the STM400 solution using Keil 5.15. Board is a V1.2 ( dont think this is relevant now) - TinyBooter and TinyCLR run fine, USB ok etc. If I run TinyCLR and monitor the network with WireShark, there are no DHCP requests etc. coming from the board. If I assign a static IP address, either using MFDeploy or from an application, board does not responds to pings. If in an application, I try to bind a socket to the interface, then an clr socket exception is thrown, with the error code Network Not Available. Testing other MF 4.3 ethernet boards in the same set up works fine, so problem seems to be in STM400 board not infrastructure.
Can anyone supply a built image ( using the MF4.4 release solution), so I can double check its not something in my build setup or infrastructure.
Anyone any suggestions on what to check to locate the problem, i.e. elimate if the problem is in the low level interface to the phy, or in LWIP or to confirm that the RTX rtos is infact running correctly - any suggestions appreciated.
A brief description of the high level architecture on how the RTX rtos is used to provide seperate threads for LWIP and CLR and how the two integrate into the CLR would be helpful.

smaillet-ms, 2016-01-11:
Hook up to uVision debugger and look in the debug printf window to see debug information. It should tell you about any errors in network startup. My guess is something is busted in the cabling or connectors and the board can't detect that a cable is connected to an active network.

SytechDesigns, 2016-01-16:
Steve, I did a clean build on the 400 project and also made it a debug build - connected to MDK debugger using a Ulink pro. When I run Tiny CLR the only debugging output I get ( apart from build date, loading mscorlib etc. and cannot find entry point ( no app loaded) is  - 
waiting for debug commands
AutoNegotiation completed with 100MB Full Duplex
Ip address from interface info 0.0.0.0
If I set a static ip address with mfdeploy, then the last line changes to the static ip address 
But there is no Ethernet - cant ping etc. and if I have dhcp enabled - there are no dhcp requests from the device.
How do I turn on debugging for lwip? or more debug messages?
Also if I open the OS System and thread Viewer window I can see the RTX threads main, tcip_thread , osTimerThread and the os_idle_demon. - 
Is it possible to send me hex files for a working image (TinyCLr) so I can eliminate some sort of build error. 
Infrastructure wise, I can run an LWIP test project on a 429eval board, and dhcp works fine. I havn't ported this to the keil board yet, but I suspect it will work - don't think its infrastructure but a problem in my build.