NETMF/netmf-interpreter #373: MCBSTM32F400 Rev 1.2 Ethernet not working

SytechDesigns, 2016-01-25:
This refers to the rev1.2 board with the KSZ8081 phy.
Builds fine, TinyClr is working. However there are no Ethernet packets being sent ( or seem to be received). 
Every thing seems correct - checking the phy registers after setup shows all are correct - the extended reg phy control 1(Eh) shows operation mode is 100MB full duplex - signal is detected on rx pair and link is up. BSR (01H) shows all is well. In eth_lwip_adaptor.cpp - network_uptime_completion is being called and the status check is showing link up.
The tcpip thread is running and in init calls dhcp_start - this formats the udp message and binds it etc. - but no message is seen by wireshark.
So the network link looks like it is up, tcpip thread is running - the dhcp state machine is sending udp messages - but no network activity is seen.  if I set a static ip address config with mfdeploy - and try to ping the static address - there is no response. I can see in lwip that the correct static address etc. has been picked up and loaded from the config.
I have written a bare bones lwip app and called dhcp start - this works and I get an ipaddress and I can ping it. I have tried 2 brand new mcd 400 boards - results the same.
'Maiorfi' sent me the image he built that works on his rev1.1 board - I tried this and the result is the same - no ethernet.
I have eliminated hardware ( 2 different boards and works with a simple lwip app), I have eliminated cables etc, The phy seems to be configured correctly - auto negotiates to 100mb full duplex, main thread is running and tcpip thread is running as are the lwip timers. - it just seems there is a problem with running it on the 1.2 board.
Has anyone got a 1.2 rev board running successfully?

smaillet-ms, 2016-01-25:
Yes, we have it running, the v4.4 code released was validated on both rev 1.1 and 1.2 boards.

SytechDesigns, 2016-01-25:
Steve - any suggestions on what I can do to get to the bottom of this? any chance of you sending me the hex files for your running image. I'm sure its just some kind of stupid set up mistake, as everything looks like it works. I have spent days on this now, every thing I look at looks like its working - just it isn't!

smaillet-ms, 2016-01-28:
Can you try with the board connected to a HUB (NOT a switch). That should put it on a half duplex mode. There were problems with the 1.1 boards on 100M links with full duplex, which is what drove changes for the 1.2 boards AFAIK. There may still be issues on some boards/networks... AS the symptoms match what we used to see on the earlier boards.

SytechDesigns, 2016-02-03:
@smaillet-ms  - 
I can get the rev1.2 board working with a simple lwip application - using the STM cube hal drivers, so I know the hardware can physically work at 100M full duplex and on my network. As all phy access in the MF phy code is just 'standard' registers I don't think the phy is configured incorrectly. I think we can assume the lwip and RTX is all working, as everything checks out, which suggests that the place I need to look is in the STM32 Mac register setup, as this is the main difference between the simple lwip app and the mf code. I will track this down, but it is on hold for the moment as we are in the middle of some higher priority LCD Touch screen porting work. - regards