NETMF/netmf-interpreter #377: SSL/TLS support?

maiorfi, 2016-01-31:
Hi.

I was playing with AMQPNetLite lib on KEIL dev board with latest sources, but whenever a AMQPS connection is made to a ServiceBus EventHub endpoint I get:

Establishing a connection...
    #### Exception System.NotSupportedException - CLR_E_NOT_SUPPORTED (1) ####
    #### Message: 
    #### Microsoft.SPOT.Net.Security.SslNative::SecureClientInit [IP: 0000] ####
    #### Microsoft.SPOT.Net.Security.SslStream::Authenticate [IP: 0051] ####
    #### Microsoft.SPOT.Net.Security.SslStream::AuthenticateAsClient [IP: 000c] ####
    #### Microsoft.SPOT.Net.Security.SslStream::AuthenticateAsClient [IP: 000a] ####
    #### Amqp.TcpTransport::Connect [IP: 00ba] ####
    #### Amqp.Connection::Connect [IP: 0011] ####
    #### Amqp.Connection::.ctor [IP: 0076] ####
    #### Amqp.Connection::.ctor [IP: 0008] ####
    #### ServiceBus.Scenarios.EventHubsExample::SendMessages [IP: 002b] ####
    #### ServiceBus.Scenarios.EventHubsExample::SendToEventHub [IP: 000f] ####
    #### ServiceBus.Scenarios.EventHubsExample::Run [IP: 0005] ####
    #### ServiceBus.EventHub.NetMF.Program::Main [IP: 0024] ####
Exception thrown: 'System.NotSupportedException' in Microsoft.SPOT.Net.Security.dll
Exception thrown: 'System.NotSupportedException' in System.Net.Security.dll

Does it actually mean that there is something missing in firmware solution building?

Thanks!

maiorfi, 2016-01-31:
After updating SSL Seed thorugh MFDeploy tool (btw, what does such action actually do?), my issue is changed into:

Establishing a connection...
    #### Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ####
    #### Message: 
    #### Microsoft.SPOT.Net.Security.SslNative::SecureWrite [IP: 0000] ####
    #### Microsoft.SPOT.Net.Security.SslStream::Write [IP: 0040] ####
    #### Amqp.TcpTransport+SslSocket::Amqp.ITransport.Send [IP: 0011] ####
    #### Amqp.TcpTransport::Send [IP: 0009] ####
    #### Amqp.Sasl.SaslProfile::Start [IP: 0087] ####
    #### Amqp.Sasl.SaslProfile::Open [IP: 0007] ####
    #### Amqp.Connection::Connect [IP: 0058] ####
    #### Amqp.Connection::.ctor [IP: 0076] ####
    #### Amqp.Connection::.ctor [IP: 0008] ####
    #### ServiceBus.Scenarios.EventHubsExample::SendMessages [IP: 002b] ####
    #### ServiceBus.Scenarios.EventHubsExample::SendToEventHub [IP: 000f] ####
    #### ServiceBus.Scenarios.EventHubsExample::Run [IP: 0005] ####
    #### ServiceBus.EventHub.NetMF.Program::Main [IP: 0024] ####
    #### SocketException ErrorCode = -1
Exception thrown: 'System.Net.Sockets.SocketException' in Microsoft.SPOT.Net.Security.dll
    #### SocketException ErrorCode = -1
    #### SocketException ErrorCode = -1
    #### SocketException ErrorCode = -1

Any idea?

maiorfi, 2016-02-01:
...trying to debug this SSL issue, I discovered that debug image is too large to be generated (more than onboard flash size).

What is the suggested way to get rid of this limitation: stubbing modules (but which ones?), moving modules on external flash or what else?

Thanks!

maiorfi, 2016-02-03:
Any news? SSL/TLS support is rapidly becoming crucial for any IoTproject! :smile: 

Thanks!

smaillet-ms, 2016-02-03:
Out of curiosity, What sort of news are you looking for here? As you know, SSL/TLS are supported via OpenSSL already. While we'd love to see a more space efficient implementation of these protocols that's a big job and isn't exactly going to happen overnight.

maiorfi-at-innovactive, 2016-02-03:
I'm sorry to be a nuisance, but I was referring to E_FAIL in SecureWrite method (2nd post).

Thanks :smile:

smaillet-ms, 2016-02-03:
The seed is a cryptographically random value used as the seed in further random value generation on a NETMF device. As most Micro controllers lack support for high entropy hardware random number generation sufficient for secure implementation of SSL/TLS this seed is generated on the desktop and written to the device.

As to the CLR_E_FAIL it looks like the same symptoms that [this commit](https://github.com/NETMF/netmf-interpreter/commit/5a37bce12a513de9f6c38f78d5cab9cab8676674) was supposed to fix. You'd need to get the debugger in there and see what's going on. There may be some subtle case of the OpenSSL platform macro use that was missed.

maiorfi-at-innovactive, 2016-02-03:
Ok, this is a good hint: I'll build image with as much stubs as possible and take a look with debugger. Thx.

maiorfi, 2016-02-04:
Hi Steve.

I spent this morning cooking a flash configuration suitable for native debugging (after inspecting TinyCLR.map I decided to put all openssl object files on external flash) and tried to follow exception raising from SslNative::SecureWrite().

Well,what I discovered is that...exception does not throw if I put a breakpoint on:

HRESULT Library_spot_net_security_native_Microsoft_SPOT_Net_Security_SslNative::SecureWrite___STATIC__I4__OBJECT__SZARRAY_U1__I4__I4__I4( CLR_RT_StackFrame& stack )

that is at line 288 of \netmf-interpreter\CLR\Libraries\SPOT_Net_Security\spot_net_security_native_Microsoft_SPOT_Net_Security_SslNative.cpp

...and let code run (F5, i.e. Go in Keil uVIsion) whenever breakpoint is hit.

This way I succeeded in publishing several events through AMQPS protocol on Azure Eventhub, definitely "...one small step for .NETMF, one giant leap for mankind" :smile: 

Is it a timing issue? Any idea about how I could fix it?

Also, during some "debugging cycles", I observed (but it could be irrelevant) that in \netmf-interpreter\DeviceCode\pal\OpenSSL\OpenSSL_1_0_0\tinyclr\ssl_write_internal.cpp, whenever 

int sent = SSL_write( ssl, data, size ); 

...returns 0xffffffff, call (just below) to

int err = SSL_get_error(ssl,sent);

...returns SSL_ERROR_WANT_WRITE, not addressed by "special" management related to...

    if(err == SSL_ERROR_WANT_WRITE)
    {
#if !defined(TCPIP_LWIP) && !defined(TCPIP_LWIP_OS)
        SOCKET_DRIVER.ClearStatusBitsForSocket( sd, TRUE );
        return 0;
#endif
    }

Thank you very much for your support!

maiorfi, 2016-02-10:
Hi. I'm really quite sorry for bothering you about this issue, but taking into consideration my latest post, is that something specific I could take a look at in order to spot the subtle cause of observed behavior (i.e. SSL/TLS working only temporaly halting native side SslStream interop calls via breakpoints)?

Thanks.