NETMF/netmf-interpreter #378: where is CLR_Defines.h?

kmshim, 2016-02-01:
hi, @smaillet-ms 
Cloned information
> 
> SHA-1: eb4596e9a599c8b62498bca22b78caa825202e0e
>* Merge pull request #354 from DerekGn/dev
> Removed assembly signing key

I know that this header file,CLR_Defines.h, had been in the \CLR\Include.
but now this file is not exist in that path.
is this header file missing? or moved to the other path?

miloush, 2016-02-01:
Looks like it needs to be built?

https://github.com/NETMF/netmf-interpreter/blob/34a92c3d79d3d1c48440dc32f88d5b81095311a4/Framework/Tools/CreateCLRDefines.proj

lt72, 2016-02-01:
the header is automatically generated during the build by this msbuild task:

GenerateCLRDefines 

it is generated into the platform independent portion of the build directory.

kmshim, 2016-02-02:
Thanks for your help.

closed 2016-02-02 by kmshim