NETMF/netmf-interpreter #382: hal for xtensa/esp8266?

nullfx, 2016-02-02:
from what i can find (which i may have missed something) the xtensa mcu on the esp8266 12 should support the netmf.  I haven't run across anyone integrating the sdk into netmf; it currently runs lua and micropython interpreters, and being a big fan of c# over lua (or python) i'd love to be able to use netmf on this mcu.

If there are no support plans for the chip i'd like to have a go at integrating its sdk into a hal for the framework, but have a lot of questions about where to start (what to cover, as the esp8266 code base is pretty extensive) / how to marry the 2 frameworks.

jeremychild, 2016-02-26:
Hey @nullfx I'd like to help you! We have lots of resources available and its something that we could do together if you'd like?

HorstBock, 2016-03-02:
Hi @nullfx and @jeremychild. I'd also like to help supporting the netmf on the esp8266. I'm very familiar with the esp8266 and the netmf. Maybe we can put our effort together. What do you think?

ishvedov, 2016-04-25:
Given that soon there will be ESP32, it would be a very good idea.