NETMF/netmf-interpreter #389: lwIP network stack does not correctly builds with GCC for 4.4

lt72, 2016-02-16:


andreaslindner, 2016-08-07:
Hi!

Does this mean that currently the lwIP stack cannot be built with GCC?

I have trouble in fixing a project in the Solution for the STM32F4-Disco board with the network extension board. To start out with, the MCBSTM32F400 TinyCLR_NET project was helpful. I adjusted my new project accordingly. But after getting it to compile and trying to run it on my board, it actually doesn't launch the CLR. With the debugging hardware and software, among other things, I found out that for some reason the SVC handler is not properly registered in the vector.

Is this not tested? Do you just use the Keil compiler for the MCBSTM32F400 TinyCLR_NET project or should GCC work as well?

smaillet-ms, 2016-08-08:
The startup and initialization sequence needs to be adapted to the GCC runtime requirements. The stack itself compiles fine. We assume it will run fine too but as of yet, no-one has stepped up to complete the GCC init work needed.

andreaslindner, 2016-08-09:
Ok, I undertand. This explains it. I assume you talk about compiling the RTX using GCC, don't you?

I'd like to produce some fix if possible, but I need a little guidance for this. Could you enumerate what are the gaps that need to be filled (as an outline)?

What I don't understand first is that the symbol SVC_handler is available in the resulting binary but it seems not to be linked in the vector table. How does this happen? Shouldn't this be independent of the toolchain as well?