NETMF/netmf-interpreter #397: Tweaks in gcc.targets for GCC 5-2015-q4-major

josesimoes, 2016-02-23:
I've been doing some tweaking in the gcc.target with the lattest GCC version.
Mostly optimizations and adding extra debug information to debug images.
Should I open a new PR for this?

smaillet-ms, 2016-02-23:
Can you be more specific on the tweaks? What happens if someone doesn't have that particular release? (e.g. will it just work?, gracefully fall back? or crash? )

josesimoes, 2016-02-23:
Mostly flags asked to optimize the outputs for "bare metal" systems that I've been collecting when going through the documentation and reading forums.
Considering that newlib nano is already separated and optional all the others are pretty straightforward, I would say. They are not exclusive of this version although the compiler flags can produce different results. In the end I don't think there'll be any catastrophic result.

smaillet-ms, 2016-02-24:
OK, sounds like something worthy of a PR without further discussion then.

closed 2016-02-25 by josesimoes