NETMF/netmf-interpreter #398: NETMF and FreeRTOS (with GCC)

josesimoes, 2016-02-23:
I've a working solution with FreeRTOS replacing RTX.
Next steps will be adding network support, including SSL.
Is this interesting enough to start a new PR? Maybe the NETMF team wants to take a look and evaluate/discuss?

smaillet-ms, 2016-02-23:
That is of interest. It Is worth discussing the general design concepts on the integration and folder layout etc.. so we can remain as consistent as possible.

josesimoes, 2016-02-23:
I think so too. This follows my repeated efforts to get RTX working with GCC. I think it's a good opportunity to move away from the Open SSL and update the IP stack in the process.
We've pretty much followed the established layout and structure. Added a folder for FreeRTOS and touched only where needed. It includes a CMSIS layer so it looks and feels very standard.
What would be the next step? I mean a PR for this would be a work in progress as there is still much to be done rewarding networking.

smaillet-ms, 2016-02-24:
Start incrementally. That is, submit a working code PR for FRE_RTOS with no networking and a simple thread that blinks an LED or something running alongside the main thread which runs the NETMF interpreter. This proves the system is running and synchronization locks are functional. Then add things one piece at a time until full networking is enabled. Don't try to do it all in one big push as that will be a real nightmare to diff and debug.

josesimoes, 2016-02-24:
That's exactly what I have right now. I'll submit a PR with that.

josesimoes, 2016-02-24:
Just submitted #402

josesimoes, 2016-02-24:
@smaillet-ms question please:

Need some guidance on how to change the libcatproj files.
For example in Sockets_PAL.libcatproj, near the end there is this entry:
  <FeatureAssociations Name="Network (LWIP OS)" Guid="{FC481D0B-784B-402B-8EE0-0BD1D29823A9}" ProjectPath="$(SPOCLIENT)\Framework\Features\Network_LWIP_OS.featureproj" Conditional="" xmlns="http://schemas.microsoft.com/netmf/InventoryFormat.xsd">
    <ComponentType>Feature</ComponentType>
  </FeatureAssociations>

should I add a new one after this, such as this?

  <FeatureAssociations Name="Network (FreeRTOS)" Guid="{D6AA9A25-4B00-476E-84B8-6A1AA2AD0BC2}" ProjectPath="$(SPOCLIENT)\Framework\Features\Network_FreeRTOS.featureproj" Conditional="" xmlns="http://schemas.microsoft.com/netmf/InventoryFormat.xsd">
    <ComponentType>Feature</ComponentType>
  </FeatureAssociations>    

Won't this - wrongly - associate both the LwIP and FreeRTOS? Or should be something in the Conditional=""?
Or should be a different Sockets_PAL.libcatproj?

TIA!

smaillet-ms, 2016-02-24:
Assuming we did our job right in v4.4 you shouldn't need to deal with the categories for LWIP, The OS variant should be good enough - it doesn't specify the particular implementation. (The whole libcatproj featureproj thing is a real mess, it started out simple enough but has rapidly grown to a real can of worms. Generalized dependency resolution, and conflict detection and prevention is a hard problem the industry as a whole hasn't really resolved)