NETMF/netmf-interpreter #417: Questions about heap(s)

josesimoes, 2016-02-29:
Looking at the scatterfiles it seems that two heap regions exist. 
One referred as heap, that I understand its the one managed by the CLR, correct? 
And there is a "custom heap" who's purpose is unclear to me. Could someone please explain it to me?

smaillet-ms, 2016-02-29:
Custom heap is, as the name implies custom to a particular firmware build. It is available to the native code implementation, and not used by the CLR. NETMF was designed at a time when the C++ runtime libraries for Micro Controllers tended to pull in the world, so it has options to handle most of the basics of a C runtime library with as little overhead as possible.

josesimoes, 2016-02-29:
Just to be 100% sure that I understood you correctly: this region is the equivalent to the regular heap region of a "non-NETMF" solution. Is this correct?

smaillet-ms, 2016-03-09:
Pretty much, yes. NETMF was designed at a time when embedded C runtimes were pretty FAT. (newlib-nano and other such micro controller bare-metal focused run-times did not exist. ) Thus it uses its own implementations for the subset of standard runtime functionality it needs. Fast forward to the year 2015 and it's a different world... Part of the HAL/PAL refactoring conversation includes leveraging existing runtime support libraries and deprecating custom implementations.

josesimoes, 2016-03-09:
:+1: 
Got it. Thanks!

closed 2016-03-09 by josesimoes