NETMF/netmf-interpreter #421: Please clarify methods to update installed SDK when developing a custom / pre-release SDK

doingnz, 2016-03-02:
What is the preferred way to update the SDK on your development machine after running build_sdk?

1. uninstall / reinstall the new SDK before building Solutions or Programs that need the updated SDK?

2. some other method to copy deploy the new files (e.g. copy from $(SPOCLIENT)\BuildOutput\... to appropriate $(XXX)\Microsoft .NET Micro Framework\v4.x\Assemblies\be and le folders.

Thank you.

smaillet-ms, 2016-03-02:
The MSI should be able to update in place without manually uninstalling (It should do the uninstall for you) but a manual uninstall/reinstall of the SDK should work as well. Though, frankly the majority of the work going on at present doesn't involve any need to rebuild or re-install the SDK so I rarely ever do that.

doingnz, 2016-03-03:
On a Windows 10 VM used for MF 4.4+ Dev I get the following if I try run the MSI without uninstall first. 
:
Windows Installer

Another version of this product is already installed.  Installation of this version cannot continue.  To configure or remove the existing version of this product, use Add/Remove Programs on the Control Panel.