NETMF/netmf-interpreter #432: how do get started for a conceptual wsn university project

OUBen85, 2016-03-23:
Hello, I'm trying to determine the best way to go about writing the code for my final project at undegraduate level study. It's a WSN, delivering humidity and temperature readings from a asphalt road. My limited programming experience is with VB. It won't be possible for me to physically implement the network but I was hoping to use visual studio and the .net micro framework to emulate the sensors and processing units and present my code in my project. The problem is that the latest netmf build doesn't seem to support express skus. Can I download an older version of netmf and run it on eg. visual studio express 2013? Any help would be much appreciated. Regards, Ben.

networkfusion, 2016-03-23:
You can use the community editions of Visual Studio, these have full support for the latest NetMF frameworks.

Also you could consider using a NetMF compatible board from a manufacturer like GHI Electronics who have done the hard task of creating the firmware, meaning you can just use VB to create the program.

OUBen85, 2016-03-23:
Brilliant. Thank you kindly for the quick reply. I'll go get downloading and will consider buying a board too.

OUBen85, 2016-03-23:
@networkfusion Would you recommend a any particular board for such a project, perhaps from past experience?

networkfusion, 2016-03-23:
I would recommend detailing your requirements on the community forum and going from there. Personally I would go for a Fez Raptor or Fez Spider II depending on what accessories you would need and the complexity of your program.

OUBen85, 2016-03-23:
Thanks again, much appreciated.

Becafuel, 2016-03-23:
@*OUBen85*  
  
Sorry for what could look like advertising but...  
You could also use a Quail board ( [http://www.mikrobusnet.org/](http://www.mikrobusnet.org/) ) with a huge number of sensors via Click boards from MikroElektronika ( [http://www.mikroe.com/click/](http://www.mikroe.com/click/)  ). MBN is providing more than 60 drivers source code for those Click boards ( [http://mikrobusnet.org/bb/index.php/board,21.0.htm...](http://mikrobusnet.org/bb/index.php/board,21.0.html) ).  
  
The hardware is all available from MikroElektronika or Mouser.  
  
  
Regards,  
_______  
Christophe  
  
@networkfusion Would you recommend a any particular board for such a project, perhaps from past experience?

    �

OUBen85, 2016-03-23:
Thank you much for the useful info.

OUBen85, 2016-03-29:
Hello again @networkfusion, I've been busy for the past few days and I'm only getting back to this now. I've just installed VS Community and netmf and everything seems to be working fine. Just to get it clear in my head, if I had a netmf compatible board I could write my program as if there were wireless sensors connected (thanks to the firmware), even though I won't actually have any wireless sensors in place?

networkfusion, 2016-03-30:
@OUBen85 You would have to provide more details as to what you are trying to achieve and I am not sure this is the correct place to do it. I would suggest using one of the forums (such as the one I suggested on GHI's website) where you can outline what you want to achieve and ask for feedback on the best way to achieve it.

If you are just trying to simulate a wireless sensor sending data to a web service, it would be a lot easier to just create a console app that sends simulated data to a web service (or what ever end point you are wanting). If you want to build a prototype wireless sensor, then a NetMF, RPI, Arduino device would be a good way forward.

OUBen85, 2016-03-31:
@networkfusion thanks again for your reply.
I was hoping to simulate wireless sensors sending data to say a laptop but I'm not sure if this is possible just by writing a program with netmf in VB. I was under the impression that maybe there were templates on netmf to help with such a project. 
Some context; The idea for my project is to have arrays of wireless sensors situated at six different points around a long race track, the Isle of Man TT course, which feed real-time humidity and temperature data through to hardware at each of those points, which then send that data to a centralised task manager within the track boundary through long range wifi, to give event organisers accurate data as to what parts of the track are damp or that have melting tar or whatever. My argument is that this would help with safety in a very dangerus motorsport event. It won't be possible for me to physically implement this network and I've been told that it's perfectly fine for me to produce a conceptual project, so I was hoping to find a way to present to the reader network topology design and the code for handling such data in my project.
Anyways, if you have any more advice based on what I've said in this post, that would be brilliant. If not, you've been very helpful and thank you kindly for your time :)

smaillet-ms, 2016-04-01:
You could do that with only the emulator if you wanted. You could create simulated data from a NETMF app running in the emulator and have it communicate (using Networking) to the host desktop. If you wanted to get to a more detailed and more production focused you could extend the emulator to actually simulate real hardware your application could then read data from the simulated hardware and pass it on to the host using the code you built in the first phase. Once you are satisfied with that, you can move to the next phase where you actually buy some hardware and run some real world tests.

OUBen85, 2016-04-01:
@smaillet-ms I was hoping I'd be able to do what you've recommended with the emulator. I'll get researching the classes etc and see what if I can figure out how to get started with creating the simulated data and go from there. Tyvm for your reply, much appreciated.