NETMF/netmf-interpreter #435: Develop .NetMF apps with non Windows platforms

AlexandrSurkov, 2016-04-12:
I have been asked about using .NetMF with Linux and MaxOS systems. I found interesting project: https://github.com/DavidKarlas/MonoDevelop.MicroFramework
But last commit was 7 month ago. As I know, Xamarin Studio can be used for Mac to create .NetMF apps. But on Linux it has some problems.

Has someone any information about plans for this addon? Since Xamarin now is part of Microsoft.

smaillet-ms, 2016-04-12:
There's also some support in the Monkey robotics project, though, again focused on Mac. There are no current plans to do any additional work to support Xamarin tools. The focus, at the moment, is delivering a great story with VisualStudio.

AlexandrSurkov, 2016-04-13:
@smaillet-ms Thanks for answer.

closed 2016-04-13 by AlexandrSurkov