NETMF/netmf-interpreter #436: NETMF 4.4 SDIO driver

dale-lyons, 2016-04-14:
Hello,
 As in my last post I am trying to develop a SDIO driver for the STM32F407 netmf 4.4
 I've run into a roadblock and I am hoping someone can suggest something I am missing
 or give me a kick in the rear.

 I am using a GHI sdcard module wire wrapped to a STM32F407 DISVOVERY board. I used the
 STMCubeX software to generate a MDK project, I build it and it works just great. Life is good Smiley
 I can initialize the card, read and write blocks etc.

 So ... I create a new driver in netmf 4.4 following the great instructions I found about netmf interop.
 The driver integrates into netmf all well and good, however the code to initialize the sdcard fails.
 The initial CMD0 works fine (go to idle) however the next command CMD8 (query voltages) fails
 with a timeout.

 When I create the CubeX version I use the exact same RCC setup that is used in the netmf bootstrap code. Again the
 cubeX version works fine and the exact same CubeX code ported to netmf fails.

 At the point netmf is performing the first access to the sdcard I check the RCC and SDIO registers and see they are
 all set correctly in each case however the netmf version still fails.

 I am completely out of ideas of what is different between the same 2 platforms. Identical hardware just a different
 startup.

 My gut tells me its a timing problem but I cannot see a difference to explain it. Both systems have a 48MHz clock
 although the netmf uses a much higher system clock than the cube version.

 Any ideas would be greatly appreciated. thanks!
 dale

cw2, 2016-04-15:
Would it be possible to look at the actual code of your NETMF driver?

dale-lyons, 2016-04-15:
sure, let me get it together and I will post it.

dale-lyons, 2016-04-16:
it looks like I had a misconfigured clock line. not sure why it works in the cubeX version
which is what misled me. thanks for the response. cheers

closed 2016-04-16 by dale-lyons