NETMF/netmf-interpreter #437: Port NETMF to NetBSD

krytarowski, 2016-04-17:
I'm willing to port NETMF to NetBSD... however I cannot find build.sh. Makefile or similar start point.

smaillet-ms, 2016-04-18:
There is no build.sh or makefile as the .NET Micro Framework is built using MSBUILD. While [MSBUILD is Open Source now](https://github.com/microsoft/msbuild), I'm not sure if there is a port for NetBSD. Assuming there is, there are still a number of custom build tasks and presumptions of a full Windows .NET framework installation in several build tasks Thus **Building** NETMF on NetBSD would be a significant undertaking at this point. However, Building on Windows to Target NetBSD is a more plausible scenario. As it only involves adapting a new cross compiler tool chain into the existing build (Which currently supports several tool chains already). While this is still a bit of work it's much less so than trying to do the build on NetBSD.

AlexandrSurkov, 2016-04-18:
I think [this documentation](https://netmf.codeplex.com/releases/view/56879) can be usefull. Especially RCLPort.chm

krytarowski, 2016-04-18:
So the first milestone is to get MSBUILD functional, I will focus on it.

smaillet-ms, 2016-04-22:
Let us know how your progress as you go. Since refactoring the build is high on the list of things to do (as part of a broader arc to make NETMF easier to use/port) I'd like to be sure we don't leave you behind when we make changes. 😄

krytarowski, 2016-05-07:
.NET CLR officially supported on NetBSD.. heading for bootstrapping native assemblies.