NETMF/netmf-interpreter #438: Compile netmf to a lib

innovak, 2016-04-18:


smaillet-ms, 2016-04-18:
NETMF builds everything to lib files already, then in the final stage it links them all together to make the executable image. PKStudio used to have some support for generating Keil Projects from the NETMF projects. There has been some recent activity regarding PKStudio recently (#429, #430 ) so you might want to check into that.
As discussed in those topics and the forward looking [WIKI article on the build system](https://github.com/NETMF/netmf-interpreter/wiki/BuildSystemRequirements) the build infrastructure and component dependency system are an area getting a lot of thought for a significant re-work so any feedback on the future thinking is welcome.

innovak, 2016-04-18:
Thank you for your reply! I will pay attention to PKStudio.

cw2, 2016-04-18:
If you manage to successfully build the firmware via the commandline (Keil toolchain support is rather good), you can create a new µVision project for the target CPU variant, set executable name to TinyCLR and browse for the appropriate directory in "Select folder for Objects" and debugging, breakpoints etc. will work as usual. You can also add the source files to the project as needed...

It is also possible to use µVision to debug GCC ELF output, just rename it to .axf.

AlexandrSurkov, 2016-04-18:
Please welcome to [PKStudio repository](https://github.com/AlexandrSurkov/PKStudio). Feel fre to ask any questions about it.

smaillet-ms, 2016-04-18:
re: @cw2 There's even a [sample uVision project](https://github.com/NETMF/netmf-interpreter/tree/dev/Solutions/MCBSTM32F400/netmfdbg) in the tree we use for the MCBSTM32F400 dev board

innovak, 2016-04-18:
Um,I understand how to debug the final .axf file with uVision,But my aim is how to integrate netmf into the existing code easily, because I do not know very much about the netmf compiler environment,So it would be great if I could use netmf easily like just onliy with some modifications of configuration files and include some .lib or other source files into my project as much as any other middleware.^-^

smaillet-ms, 2016-04-18:
NETMF has no specific support for integrations into any given IDE for firmware builds (sadly note even VS at this point, but that's something I'm working to change). Every IDE uses their own custom project and build systems, there's no single universal representation of projects. Thus some manner of conversion tool is required. PKStudio is one such tool getting some work of late. If you know the format of the IDE project files you are targeting there's likely a way to auto generate an appropriate project to bring it all together. While there's nothing we can do to make a universal project useable as-is in all IDEs we can, and should do  a lot more to make the process of building NETMF a lot simpler, which hopefully would also make mapping/converting to various IDEs simpler. The current design is based on market realities that simply don't exist anymore.

closed 2016-05-14 by smaillet-ms