NETMF/netmf-interpreter #440: Advice for adding FreeRTOS for non CMSIS cpu

DerekGn, 2016-04-21:
Hi

I have just started implementing network support on the SAMA5D3 explained hardware. I am looking for some guidance on where to add the necessary support files for FreeRTOS on a non CMSIS compliant cpu.

There is currently a folder $(SPOCLIENT)\DeviceCode\Targets\OS\CMSIS_RTX

Would $(SPOCLIENT)\DeviceCode\Targets\OS\FreeRTOS be the correct place to put FreeRTOS and $(SPOCLIENT)\DeviceCode\Targets\OS\CMSIS_RTOS\DeviceCode\lwip_1_4_1_os for the lwip files like cc.h etc

smaillet-ms, 2016-04-22:
No FreeRTOS code should be in the NETMF tree itself. We currently have a placeholder directory for the CMSIS code to be dropped in, but that was a bad idea I intend to revert. (To better allow use of CMSIS-Pack and an existing installed repository) The only things that should be in the tree is the code to act as an adapter between the underlying platform and NETMF. Generally the best place for that is under $(SPOCLIENT)\DeviceCode\Targets\OS\<OS name>

DerekGn, 2016-04-22:
Ok thanks I will follow this advice

closed 2016-04-22 by DerekGn