NETMF/netmf-interpreter #442: Debug protocol

josesimoes, 2016-04-24:
Is there any documentation on the debug protocol and/or how communication is supposed to work between MFDeploy/VS and a NETMF device?
I would like to have a better understanding on how that works because it would help me on debugging the USB CMSIS driver.
At a first look that code in the MFDeploy project looks more like a port from C++ to C# and it doesn't use any of the modern stuff available in terms of USB APIs, async/await, etc.

smaillet-ms, 2016-04-25:
No docs available, Though it's a simple packet based protocol (with one annoying exception). The implementation on the desktop - no so much with the simple. 😉 It was written in the very early days of NETMF development (circa .NET 1.0) so, as you point out has no modern support. I recently began experimenting with re-writing MFDeploy as a Universal app (So I could better understand the APIs available as well as see about defining a PCL for NETMF device communications. While the USB device APIs are pretty decent - it isn't without issues. The biggest is that you must know the VID+PID of the device(s) your app will talk to at build time (e.g. the app must declare what it intends to talk to in it's manifest). I'm trying to find someone I can talk to about why that is true even for a WinUSB device with a defined InterfaceGuid.

The actual packets are all defined as "commands" in the code. In v4.4 I added tracing support with some basic packet parsing into the desktop side code base, you can see the results in the VS Debug window so I could diagnose debug issues as well. I also added tracing into the PAL implementation though you need to enable that in the code and re-build (as it adds footprint due to extra code and strings I didn't want it on all the time). To enable the PAL tracing you can set TRACE_MASK to include the TRACE_STATE and or TRACE_HEADERS flags ins[ WireProtocol.cpp](https://github.com/NETMF/netmf-interpreter/blob/a35ee529ce7930b31873d1524e9f44db1d4bbed3/Support/WireProtocol/WireProtocol.cpp)

The one REALLY annoying exception is the desktop receiver must handle arbitrary data not conforming to a packet as valid debug message strings. (e.g. the device can send arbitrary strings without the standard header+payload+footer envelope, and the receiver treats such data as a debug message string. (Despite the fact that there is an actual debug message packet defined) That one little gem complicates the heck out of the desktop side code base.

josesimoes, 2016-04-25:
Thank you for your explanations.

Regarding the requirement for a UWP to declare the devices it intends to communicate with I can tell you that you have no way out of that. It started with the new APIs of Windows 8.0 and that is still valid for all version after. The official explanation is security and privacy. The user must know beforehand the app capabilities and requirements AND it has to give it's explicit permission for that to happen. No way around that.

As for the protocol in starting to get my head around it. Although its a single packet protocol it has a lot of extra complications that seem a bit to much...
Right now I'm having a peculiar and annoying issue: I have to add a fake delay of ~200ms after each USB transmission in order to have MFDeploy to be able to talk with the device....

smaillet-ms, 2016-04-25:
Fake delay... interesting.  You can get some protocol info out of a debug build of MFDeploy either by using the VS debugger, or by [modifying the application config file](https://msdn.microsoft.com/en-us/library/sk36c28t(v=vs.110).aspx) to see the wireprotocol trace messages that I added in v4.4 (see: [DebuggerEventSource.cs](https://github.com/NETMF/netmf-interpreter/blob/dev/Framework/Debugger/DebuggerEventSource.cs) )

josesimoes, 2016-04-25:
Indeed it is. If you happen to remember how it works (on the device side) it tx a response packet is two "batches" the first with the header and the second with the payload. With this driver if the follows the expected workflow MFDeploy never catches the payload only the first batch. The second part remains lingering in the USB layer until the next "ping" when it shows as being the first packet being read. And this goes on.
As soon as I insert that fake delay it starts working. 

This is what caused me to look into the protocol from the host side trying to find out what was happening.