NETMF/netmf-interpreter #444: Undefined build variables $(IntermediateOutputPathLE) & $(IntermediateOutputPathBE)?

doingnz, 2016-04-26:
Please see line 660 and 722 in ```tools\Targets\Microsoft.SPOT.Build.Targets```

```
660     <CreateProperty Value="$(IntermediateOutputPathLE)TinyCLR_DebugReferences.cache"> 
661       <Output TaskParameter="Value" PropertyName="TinyCLR_DebugReferencesStateFile" /> 
662     </CreateProperty> 
...
722     <CreateProperty Value="$(IntermediateOutputPathBE)TinyCLR_DebugReferences.cache"> 
723       <Output TaskParameter="Value" PropertyName="TinyCLR_DebugReferencesStateFile" /> 
724     </CreateProperty> 

```

I cannot find where ```$(IntermediateOutputPathLE)``` & ```$(IntermediateOutputPathBE)``` are defined?

It appears they are blank and multiple instances of the file ```TinyCLR_DebugReferences.cache``` are then written to the various project source folders. 

I would have expected copies of this file to be under```BuildOutput```.

Their presence in the source tree makes comparisons for changes unnecessarily difficult. 

Is it correct to simply drop the LE and BE from the variable name and use the general ```$(IntermediateOutputPath)``` ?

Things appear to work correctly and the .cache files are written to project specific paths. e.g.
```
BuildOutput\Public\Release\Client\obj\MFUpdate\__COMMON
```

doingnz, 2016-04-26:
a related concern is why the *.feedback files are being written/copied to ```\tools\make\feedback\``` ?
e.g.lines 81, 82, 102, 324 & 325 in tools\Targets\Microsoft.Spot.system.mdk.targets

```
    <CC_CPP_COMMON_FLAGS Condition="'$(reducesize)'!='true'">$(CC_CPP_COMMON_FLAGS) $(SWTC)feedback $(SPO_SDK)\tools\make\Feedback\$(TARGETPLATFORM)_$(COMPILER_TOOL_VERSION).feedback</CC_CPP_COMMON_FLAGS>
    <CC_CPP_COMMON_FLAGS Condition="'$(reducesize)'=='true'">$(CC_CPP_COMMON_FLAGS) $(SWTC)feedback $(SPO_SDK)\tools\make\Feedback\$(TARGETPLATFORM)_$(COMPILER_TOOL_VERSION)_loader.feedback</CC_CPP_COMMON_FLAGS>
...
    <LINK_FLAGS>$(LINK_FLAGS) $(SWTC)no_keep_init_arrays $(SWTC)feedback $(BIN_DIR)\$(AssemblyName)_$(TARGETPLATFORM)_$(TARGETLOCATION)_$(FLAVOR)_$(COMPILER_TOOL_VERSION).feedback</LINK_FLAGS>
...
    <Exec Condition="'$(AssemblyName)'=='TinyCLR'"    Command="copy /Y $(BIN_DIR)\$(AssemblyName)_$(TARGETPLATFORM)_$(TARGETLOCATION)_$(FLAVOR)_$(COMPILER_TOOL_VERSION).feedback $(SPO_SDK)\tools\make\Feedback\$(TARGETPLATFORM)_$(COMPILER_TOOL_VERSION).feedback" ContinueOnError="true"/>
    <Exec Condition="'$(AssemblyName)'=='TinyBooter'" Command="copy /Y $(BIN_DIR)\$(AssemblyName)_$(TARGETPLATFORM)_$(TARGETLOCATION)_$(FLAVOR)_$(COMPILER_TOOL_VERSION).feedback $(SPO_SDK)\tools\make\Feedback\$(TARGETPLATFORM)_$(COMPILER_TOOL_VERSION)_loader.feedback" ContinueOnError="true"/>

...

```

I would have expected these to remain under the BuildOutput folder. e.g.  ```$(IntermediateOutputPath) ``` in place of ```$(SPO_SDK)\tools\make\Feedback\```. Would need to also update prop_feedback_files.cmd.

smaillet-ms, 2016-04-29:
I'll need to look into the LE/BE intermediate path - not sure what's going on there. 

The feedback files and their location is a weird hold over from when we shipped an actual Porting Kit, instead of being Fully Open Source. The feedback files were pre-generated and included in the PK, so that solutions could leverage them for builds. Though the value was somewhat limited by the fact that everything builds as static libs and then gets linked together in the end. The feedback files are used in a multi-pass compilation and don't really help a lot if the libraries aren't rebuilt. (e.g. the idea is that the linker keeps notes on what was actually used during the link phase and feeds that back to the compiler so that it can re-compile the code, leaving out the parts that the linker told it aren't going to be used. Then the linker kicks in and, possibly finds more stuff unused, and the process repeats up to some fixed cycles or until the set of items unused is empty or identical to the previous pass (meaning the compiler can't do any better than it has). In the future updated build, I'd agree the feedback files should be outside of the source tree. (As with any other files generated during the build). Ideally you could put the source onto a CD-ROM, set your build output to a RAM drive and successfully build.

doingnz, 2016-05-09:
Thank you for explanation re feedback files.

It would be great to move the build system to isolate the source folders from all build output. 

Commercial systems will usually need to fork NetMF into a private repository as they have portions of code that is proprietary, or there are other aspects forcing a different version control system, etc. In essence it becomes necessary to manually maintain independent copies of NetMF (public & private) in different repositories and manually transfer changes between them. 

We should aim to make large scale comparisons of folders and files using automated tools as easy as possible. Then it is easier to mange the diff process between the images. e.g. removing "noise" from build generated cache or .feedback files that clutter the output. Easier then for commercial users to contribute improvements to the community. 
 
I have seen no obvious adverse effects from removing the LE and BE as suggested in the first post.