NETMF/netmf-interpreter #447: Support multiple I2C

techcap, 2016-05-10:
Now MF looks supporting only one I2C.
How can I use multiple I2C easily?

josesimoes, 2016-05-10:
I think that from the very beginning NETMF only supports a single I2C bus...

techcap, 2016-05-10:
Now my idea is like this.

Change I2C_Internal_GetPins() implementation to get the pin information from global static variable.
Add Interop dll for changing pin information.

1. Set pin information through Interop dll.
2. Create I2CDevice

3. Change pin information for second I2C.
4. Create I2cDevice for second I2C.

Will it be working properly?

josesimoes, 2016-05-10:
It's more complicated than that. You need to initialize, configure and manage a second I2C module (if available in the platform).

Is there any hardware (or other) reason that prevents you from using I2C addressing to deal with multiple devices on the same bus?

techcap, 2016-05-10:
1. When I2CDevice is created, Initialize() will be performed. So I thought it will be no problem. Did I miss something?

2. Because of some device which cannot change I2C address and I have to use that two devices. T.T

smaillet-ms, 2016-05-14:
The new Windows.Devices namespace is designed to accommodate multiple controller, however since the current NETMF HAL/PAL APIs don't consider that possibility it is currently limited to just one. This is something we'll be looking into for the future HAL/PAL API layering as we should not arbitrarily limit the number of any particular type of device except those that are architecturally defined (i.e. one interrupt controller for a Cortex-M based SoC)