NETMF/netmf-interpreter #460: Help to run netmf as a interpreter

yindongfei, 2016-05-26:
I want to use netmf-interpreter as a script interpreter(C# language) integrate into my game engine, I have compile all the source and make a test.pe using MetaDataProcessor.exe, when I want to load the test.pe using 

MinMFWin32.exe -load test.pe

the output like this:
```
Loading PE File: C:\Users\dongfei\Documents\Visual Studio 2013\Projects\MFConsoleApplication1\bin\Debug\test.pe

TinyCLR (Build 4.4.0.0)

Starting...
Created EE.
Started Hardware.
Loading Deployment Assemblies.
Attaching deployed file.
   Assembly: test (1.0.0.0)   (260 RAM - 736 ROM - 284 METADATA)

   AssemblyRef    =        4 bytes (       1 elements)
   TypeRef        =       52 bytes (      13 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       48 bytes (      12 elements)
   TypeDef        =        8 bytes (       1 elements)
   FieldDef       =        0 bytes (       0 elements)
   MethodDef      =        4 bytes (       2 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =      309 bytes
   Signatures      =       16 bytes
   ByteCode        =       13 bytes

Resolving.
Link failure: some assembly references cannot be resolved!!

Assembly: test (1.0.0.0) needs assembly '' (3.0.7186.0)
Error: a3000000
```

the source file of test.pe is like this:
```csharp
namespace MFConsoleApplication1
{
    public class Program
    {
        public static void Main()
        {
            int i = 3;
            i += 4;
        }
    }
}
```

then I don't know how to make it work

smaillet-ms, 2016-05-26:
At a minimum it will need mscorlib.pe. The blank name is worthy of a bug though. NETMF assumes all PE "files" are locatable in memory and doesn't have any form of automatic dynamic loading. So you'll need to specify all the dependent PEs Or modify the MinMFWin32.exe to suit your needs (You could use the existing libraries built into NETMF to extract the dependencies from the PE file once loaded and then find and load them before starting the main interpreter loop. That's not something we've had any need for as the general goal of MinMFWin32 was to get as close to a real micro-controller based device experience as possible on a Win32 app hosting NETMF so we can test/debug more effectively.

yindongfei, 2016-05-27:
I modified MinMFWin32.exe to load mscorlib first then test.pe, like the output shows, I think the test.pe file I generrated is not valid, when I attach the debugger to MinMFWin32.exe, I found the 
parameters passed into function
```c++
bool CLR_RT_Assembly::FindTypeDef( LPCSTR name, LPCSTR nameSpace, CLR_RT_TypeDef_Index& idx )
```
is wrong, the caller pass the nameSpace first and then type name, but the callee need type name first, then nameSpace.

So, I guess the way I generate the test.pe in a wrong way, I try to use then MetaDataProcessor.exe compiled by myself to generate the test.pe, but I got an error like:
```
C:\Users\dongfei\Documents\MFConsoleApplication1\bin\Debug>MetaDataProcessor.exe -loadHints mscorlib mscorlib.pe -parse test.exe -comiple test.pe
MMP: error MMP0000: 0x8013110e
```
then what should I do next? 

```
Loading PE File: mscorlib.pe
Loading PE File: test.pe
Cannot open 'test.pe'!
Error loading PE File (0xff000000 ): test.pe

TinyCLR (Build 4.4.0.0)

Starting...
Created EE.
Started Hardware.
Loading Deployment Assemblies.
Attaching deployed file.
   Assembly: mscorlib (4.4.0.0)   (4344 RAM - 40352 ROM - 22414 METADATA)

   AssemblyRef    =        0 bytes (       0 elements)
   TypeRef        =        0 bytes (       0 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =        0 bytes (       0 elements)
   TypeDef        =     1160 bytes (     145 elements)
   FieldDef       =      300 bytes (     149 elements)
   MethodDef      =     1828 bytes (     914 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =       16 bytes (       4 elements)
   Resources       =      232 bytes (      29 elements)
   Resources Files =       16 bytes (       2 elements)
   Resources Data  =      437 bytes
   Strings         =      963 bytes
   Signatures      =     3102 bytes
   ByteCode        =    16126 bytes

Attaching deployed file.
   Assembly: test (1.0.0.0)   (260 RAM - 736 ROM - 284 METADATA)

   AssemblyRef    =        4 bytes (       1 elements)
   TypeRef        =       52 bytes (      13 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       48 bytes (      12 elements)
   TypeDef        =        8 bytes (       1 elements)
   FieldDef       =        0 bytes (       0 elements)
   MethodDef      =        4 bytes (       2 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =      309 bytes
   Signatures      =       16 bytes
   ByteCode        =       13 bytes

Resolving.
Resolve: unknown type: get_Angle.System
Error: ff000000
```

smaillet-ms, 2016-05-29:
Turn on full logging in VS, build your app using VS and the SDK targeting the emulator. You can then open the log files to see the detailed command line actually executed by the build. There are a number of arguments and options to MetadataProcessor, including the "endianness" of the target device that must be set correctly.

yindongfei, 2016-06-08:
I am sure using the exactly same command line args with VS used like this:
```
C:\Program Files (x86)\Microsoft .NET Micro Framework\v3.0\Tools\MetaDataProcessor.exe -loadHints mscorlib "C:\Program Files (x86)\Microsoft .NET Micro Framework\v3.0\Assemblies\mscorlib.dll" -parse "C:\Users\dongfei\Documents\Visual Studio 2013\Projects\MFConsoleApplication1\obj\Debug\test.exe" -minimize -compile "C:\Users\dongfei\Documents\Visual Studio 2013\Projects\MFConsoleApplication1\obj\Debug\test.pe
```
when change the MetaDataProcessor.exe, nothing generated.

I build the whole project follow this:
[https://github.com/NETMF/netmf-interpreter/wiki/Getting%20Started](url)

What should I do to make it work?

yindongfei, 2016-06-08:
finally I revert all changes and rebuild whole project, and make sure all the arguments are same as VS, and it works.

thank you,

then I will try to remove the HAL and make it work as an interpreter only.

closed 2016-06-08 by yindongfei