NETMF/netmf-interpreter #462: Flash Sector Map Command - were is phantom File System section coming from?

doingnz, 2016-05-30:
Any thoughts where the "phantom" section is coming from?

If I have an application deployed to my Meridian Hardware (iMXS), TinyCLR reports the Flash Sector map as follows (via MFDeploy). 
```
Pinging... TinyCLR
Flash Sector Map Command
 Sector    Start       Size        Usage
-----------------------------------------------
    0    0x10000000  0x00010000   Bootstrap
    1    0x10010000  0x00020000   Bootstrap
    2    0x10030000  0x00170000   Code
    3    0x101a0000  0x00630000   Deployment
    4    0x107d0000  0x00010000   EWR Storage (A)
    5    0x107e0000  0x00010000   EWR Storage (B)
    6    0x107f0000  0x00010000   Configuration
    7    0x00000000  0x76100000   File System
Flash Sector Map Complete
```
Note that 0x00000000 is SDRAM on my system after memory initialization after reset and launch of TinyCLR. (1 MB SDRAM at 0x0000-0000 double mapped from the last 1MB of the physical SDRAM range 0x08000000). 

Flash is present from 0x10000000 to 0x107FFFFF (i.e. 8MB) 

If I erase the application and reset the board (or reboot TinyCLR) it drops the last sector 7 above (File System)

```
Pinging... TinyCLR
Flash Sector Map Command
 Sector    Start       Size        Usage
-----------------------------------------------
    0    0x10000000  0x00010000   Bootstrap
    1    0x10010000  0x00020000   Bootstrap
    2    0x10030000  0x00170000   Code
    3    0x101a0000  0x00630000   Deployment
    4    0x107d0000  0x00010000   EWR Storage (A)
    5    0x107e0000  0x00010000   EWR Storage (B)
    6    0x107f0000  0x00010000   Configuration
Flash Sector Map Complete
```

my flash map is defined as follows. No UPDATE, SIMPLE A and SIMPLE B regions defined.

```
const BlockRange g_Meridian_Flash_BlockRange_Region1[] =
{
    { BlockRange::BLOCKTYPE_BOOTSTRAP  ,  0,  7 },   // TinyBooter
};

const BlockRange g_Meridian_Flash_BlockRange_Region2[] =
{
    { BlockRange::BLOCKTYPE_BOOTSTRAP  ,  0,  1 },   // TinyBooter
    { BlockRange::BLOCKTYPE_CODE       ,  2, 24 },   // TinyCLR runtime
	{ BlockRange::BLOCKTYPE_DEPLOYMENT,  25,      59 + 64 },   // Application Deployment
	{ BlockRange::BLOCKTYPE_STORAGE_A,   60 + 64, 60 + 64 },
	{ BlockRange::BLOCKTYPE_STORAGE_B,   61 + 64, 61 + 64 },
	{ BlockRange::BLOCKTYPE_CONFIG,      62 + 64, 62 + 64 }    // g_ConfigurationSector
};

#endif

const BlockRegionInfo  g_I28F320_C3_16_BlkRegion[Meridian_Flash__NUM_REGIONS] = 
{
    // First 64K of flash is divided into 8 erasabe blocks 8K in size
    { 
        FLASH_MEMORY_Base,		  // Start
        BlocksInRegion1,          // NumBlocks
        BytesPerBlockInRegion1,   // BytesPerBlock
        ARRAYSIZE_CONST_EXPR(g_Meridian_Flash_BlockRange_Region1),  // NumBlockRanges
        g_Meridian_Flash_BlockRange_Region1,						  // BlockRanges
    },
    // rest of the flash is made of 63 blocks that are 64K each
    { 
        (FLASH_MEMORY_Base + (BytesPerBlockInRegion1 * BlocksInRegion1)),  // Start
        BlocksInRegion2,                            // Total number of blocks in this region
        BytesPerBlockInRegion2,                     // Number of bytes per block
        ARRAYSIZE_CONST_EXPR(g_Meridian_Flash_BlockRange_Region2),  // NumBlockRanges
        g_Meridian_Flash_BlockRange_Region2,						  // BlockRanges
    },
};
```

doingnz, 2016-05-30:
To be clear this is happening on my port to NetMF 4.4 running on the ARM920T processor, Meridian hardware (ex Device Solutions).

Interestingly it also occurs when I load my older 4.1 (with some changes from 4.2 ~ 4.4 merged)

smaillet-ms, 2016-05-31:
What version of MFDeploy are you using? Does it behave the same on older versions? (e.g. Can we isolate the problem to the desktop side or the device?)

SytechDesigns, 2016-05-31:
This may or may not be connected, but having vast experience with the Meridian there was a problem towards the end of production. Original flash went obsolete, so was replaced. marting re jigged the flash driver so it would work with boards with both flash types. After this change ( and also you may have the 200Mhz version of the old iMXS processor) , commissioning a module became a bit of a problem. After loading tiny booter, device did not always come up. Basically you had to repeatedly hit the reset and eventually it would manage to get all the way thru boot up. Strangely, once it did this once, it worked fine afterwards. We never got to the bottom of this, but it was always put down to problems with the new flash driver, maybe a combination of the faster processor and new flash. But we always suspected some sort of hidden problem with the driver and maybe a problem with the config section - as on a new bootup this would be 'empty' so this would be detected and default config data loaded - This initial loading of the config data was the only 'new device' bootup' function that changed from first boot to subsequent boot.
As I say - most likely nothing to do with your problem - but maybe there is a connection

doingnz, 2016-05-31:
Yes we are running to 200 MHz version and have the flash driver that can do both AMD and Intel flash types.

On boot the Meridian version of the hal will look for a custom section in the config used to remember which device to use for debug/NetMF and another for the settings (LCD orientation, size, clocks, etc) If not present, defaults would be defined and stored in the config. We did improve this code for NetMF 4.1 some time back.

For NetMF 4.4 my intention is to side step this initially and use default values in the app. I will review the code to check reality matches my intentions :-)

doingnz, 2016-12-08:
Mystery solved.

When running TinyCLR, the system is detected the inserted SD card. This calls ```BlockStorageList::AddDevice``` to add the SD file system.

If I remove the SD card, then execute the Flash Sector Map Command, then the expected Flash blocks are listed. 

As my code is derived from Device Solutions NetMF 4.1, need to check if the approach is still valid under NetMF 4.4

Is it correct that I add the SD card as a Block Storage Device as per the following code when an SD Card is inserted?

Or do I modify the code to initialize the driver as would have been done inside AddDevice, and then call FS_MountVolume(...) and avoid it being added as a block storage device? Wider implications to other parts of NetMF?  

If we can confirm the code is doing the right thing, then we can close this issue off. 

```
void SD_InsertIsr( GPIO_PIN Pin, BOOL PinState, void* Param )
{
	// Check for card insertion
	if( SDCardIsInserted() )
	{
		BOOL success = BlockStorageList::AddDevice( 
			/*BlockStorageDevice*/ &g_SD_BS, 
			/* IBlockStorageDevice*/ &g_SD_BS_DeviceTable, 
			/*SD_BLOCK_CONFIG*/ &g_SD_BS_Config, 
			/*init*/ TRUE );
      
		if (success)
    {
      FS_MountVolume( 
        /*nameSpace*/ "SD1", 
        /*serialNumber*/ 0, 
        /*deviceFlags*/ 0, 
        /*blockStorageDevice*/ &g_SD_BS ); 
    }
	}
	else // card has been removed
	{
		FS_UnmountVolume( /*blockStorageDevice*/ &g_SD_BS );
		BlockStorageList::RemoveDevice( /*blockStorageDevice*/ &g_SD_BS, /*uninit*/ TRUE );
	}
	
}
```