NETMF/netmf-interpreter #466: Describe using build_solution.cmd to build solutions

doingnz, 2016-06-10:
It would be nice to add some notes to Getting Started section of the wiki that points to using the build_solution.cmd.

This has the immediate advantage that setlocal/endlocal are used to avoid cluttering the cmd.exe Environment and accidentally building a second solution with left over settings.

I use a slightly modified build_solution.cmd to ensure that version numbers and time stamps in files are updated when a new build is created. I also add the MEMORY and FLAVOR_DAT to the log name.

```
 @echo off
setlocal
call setenv_%1 %2 
rem ensure visible version number is updated.
.\tools\bin\touch.exe .\Solutions\%2\TinyCLR\tinyclr.cpp  
.\tools\bin\touch.exe .\Solutions\%2\TinyBooter\TinyBooterEntry.cpp
.\tools\bin\touch.exe .\Application\TinyBooter\TinyBooter.cpp
.\tools\bin\touch.exe .\DeviceCode\Initialization\tinyhal.cpp 

cd Solutions\%3
msbuild /flp:verbosity=detailed /clp:verbosity=minimal /fileLogger /fileLoggerParameters:LogFile=msbuild_sln_%FLAVOR_MEMORY%_%FLAVOR_DAT%.Log

endlocal
```

a further note to be added to suggest that the cmd.exe used to run build_sdk.cmd is discarded prior to running build_solution.cmd to start and maintain a "clean" environment. (build_sdk.cmd will have run setenv_vs.cmd)

doingnz, 2016-06-10:
Also, it should be noted that the output from the build process can be different between the first and subsequent builds due to the feedback files created during the first run informing the compiler's optimization during subsequent builds.

The output should be equally functional, but the second run can be slightly smaller as unused portions are discard. Relevant if one is trying to compare the output from two builds (or different tools )

smaillet-ms, 2016-06-10:
build_sdk.cmd really needs to do the same Setlocal/endlocal to prevent bleeding. Ideally, when the build system is re-worked there won't be any of these sort of Environment variables or setenv_xxx etc...

Yeah the feedback files are a weird thing, they are basically a crude form of Link time optimization . The uvision tools will run the compile/link cycle in a loop until it detects no further reductions possible. The NETMF build only runs the cycle once. Unfortunately that multipass cycle isn't very compatible with common build systems (e.g. MSBuild or basic makefiles don't have any way to cleanly deal with that)

doingnz, 2016-06-10:
For future reference, MFTestSystem will need quite a bit of work to get rid of its dependence on environment variables. The code is sprinkled with references to environment variables, which in some cases, but not all, have a fall back to the system registry.

I discovered this when I tried to make the solution for MFTestSystem run using BuildEnv.props imported during the sln/project load. A fix might be to firstly consolidate all the settings to a common (static?) class all parts of MFTestSystem references, and then populate that from BuildEnv.Props.

For now, it is necessary to run it in a cmd.exe with SPOCLIENT and a few other environment variables defined (unless you are using the Test SDK, in which case it uses the registry.)

smaillet-ms, 2016-07-09:
Yes, the general idea for the build was that for those building the product itself you used environment variables set up by setenv_xxx so that you could run/debug things without doing an install first. While I've never been a fan of the setenv type scenarios (in fact I'm outright hostile to it!) there wasn't much in the way of alternative at the time this was all created. With major improvements in VS there's not much left that requires such a thing and we can solve the problem in other ways. 

The VS15 common project system is finally unifying the managed and unmanaged projects andextensibility mechanisms so we'll be able to sue that as the basis for the build and project systems for NETMF. (See: [Roslyn Project System](https://github.com/dotnet/roslyn-project-system) and [VS Project System](https://github.com/microsoft/vsprojectsystem) ). [I've been waiting for that convergence for a while now... ]

cw2, 2016-07-12:
@smaillet-ms Regarding the improvements of project systems, do you already have anything in the work queue? I have been waiting for that for quite some time too, I have already looked at some examples (nuproj, Visual C++ for Linux Development extension etc.) and possible ways to use them for NETMF build system. I would like to gather some general ideas about the new build system and hopefully start developing a proof-of-concept.