NETMF/netmf-interpreter #467: Add Task/async/await to NETMF

josesimoes, 2016-06-13:
(ongoing discussion in the [Gitter room](https://gitter.im/NETMF/netmf-interpreter) )

osre77, 2016-06-13:
generics are a way higher priotity to get in NETMF for me

josesimoes, 2016-06-13:
@osre77 generics support would be awesome! 
There is an open issue for this: #281. You may want to add a vote there or post a comment.

smaillet-ms, 2016-06-13:
Generics is essentially a pre-requisite for async/await. So is the entire System.Threading.Task.... library (which is huge) and can't just be pulled over from the desktop/CoreCLR. One of the things we found when getting async/await to work in Llilum was that the desktop implementations have some really complex optimizations that have deep implicit understanding of the garbage collector and JIT environment. None of which is appropriate for NETMF.  Though the Llilum porting could prove useful, without generics you are pretty limited on what you can do with async/await. 

Thus getting to async/await as it would be expected by most developers is a fairly large scale item.

josesimoes, 2016-06-13:
Understood. What about a 'Generic' generic implementation returning object (or a subset of the supported types). The awaiter would be responsible for casting to the correct type. Would this ease or simplify the implementation?

smaillet-ms, 2016-06-13:
While that might be a useful internal stepping stone, it isn't something that would ever make it into a public release. The relationship and dependencies between Task and Task<T> are too tight (i.e. How do you create a completed Task without Task.FromResult<>() etc...) We have looked at this a lot. Initially for NETMF 4.4 and then for Llilum. There's really no happy solution where async/await work in enough cases to make a release out of it where we don't also need generics. That's why generics is the first step in the whole shebang. (And it's why I've been documenting the internal structures of the PE files, and soon more of the run-time internals - so that I have all the major info documented to enable implementing generics). 

It is worth noting that generics, while a useful coding technique. will likely incur a performance hit due to the nature of an interpreter. Since generic types effectively means multiple type lookups to resolve the correct type of a value.