NETMF/netmf-interpreter #471: Mismatch between struct MediaAttribute and BlockDeviceInfo init in Solutions?

doingnz, 2016-06-26:
BlockStorage_decl.h defines 
```
struct MediaAttribute
{
    BOOL Removable :1;
    BOOL SupportsXIP:1;
    BOOL WriteProtected:1;
    BOOL SupportsCopyBack:1;
    BOOL ErasedBitsAreZero:1;
};
```
The Solutions do not explicitly set ErasedBitsAreZero.

eg: for STM32F4DISCOVERY
```
const BlockDeviceInfo g_STM32F4_DeviceInfo=
{
    {  
        STM32F4__IS_REMOVABLE,             // BOOL Removable;
        STM32F4__SUPPORTS_XIP,             // BOOL SupportsXIP;
        STM32F4__WRITE_PROTECTED,          // BOOL WriteProtected;
        STM32F4__SUPP_COPY_BACK            // BOOL SupportsCopyBack
    }
...
```

Need 
```
#define STM32F4__ERASED_BITS_ARE_ZERO  FALSE
```
And
```
const BlockDeviceInfo g_STM32F4_DeviceInfo=
{
    {  
        STM32F4__IS_REMOVABLE,             // BOOL Removable;
        STM32F4__SUPPORTS_XIP,             // BOOL SupportsXIP;
        STM32F4__WRITE_PROTECTED,          // BOOL WriteProtected;
        STM32F4__SUPP_COPY_BACK,           // BOOL SupportsCopyBack
		STM32F4__ERASED_BITS_ARE_ZERO      // BOOL ErasedBitsAreZero
    },
...
```

cw2, 2016-06-26:
Hm, I would be very surprised if the 'missing' bit field was not automatically set to zero, due to the rules that apply for partial initialization (starting point for C/C++ standard references at http://stackoverflow.com/a/10828333).

doingnz, 2016-06-26:
Yes, thankfully the default value of zero is OK.

Explicit initialization would show a conscious decision  was made. And its a heads up to anyone cloning the solution.

smaillet-ms, 2016-07-09:
That's trivial enough - feel free to submit a PR that includes setting that.