NETMF/netmf-interpreter #473: Remove "custom heap"

josesimoes, 2016-07-04:
Considering that:
- modern compilers/runtimes are more than capable of handling memory management
- the use of newlibnano require that an implementation of sbrk and other system calls is provided (see https://sourceware.org/newlib/libc.html#Syscalls)

I suggest that the low level memory management (custom heap) should be removed from NETMF and replace it with the "standard" memory management support.

PS: this has already been addressed in #417 not to mention that it looks pretty obvious considering what exists today.
So here it is the issue before the PR.

smaillet-ms, 2016-07-09:
👍

josesimoes, 2016-07-13:
Is this a yes? ☺ 

This, along with other improvements, would address the end symbol being discussed oh the that other PR.

smaillet-ms, 2016-07-13:
Yes, the thumbs up is an approval. I've always hated that NETMF did it's own thing instead of using a standard compilers run time library. Unfortunately when NETMF was first created those libraries were quite bloated and not well tuned for embedded micro controller use. Now with MDK uLib and newlib-nano it's a very different world.