NETMF/netmf-interpreter #479: mdk compiler libs

SytechDesigns, 2016-07-16:
In the mdk target file for MF4.4, it adds support for compiler versions above 4.x - i.e 5.x . 
If I understand the xml correctly - it looks like if the mdk version is below 5.x ( i.e 4.x) then the compiler libs are set to ARM\RV31 - but if 5.x onwards they are set to ARM\ARMCC\lib. The RV31 libs still exist in mdk5 - so the change must be for a reason - are these compiler libs 'better'?

cw2, 2016-07-16:
The RV31 libs are most likely left-over from previous installations or were installed via a "Legacy Support" package, in MDK release 5.10 (which contains ARM compiler 5.04u1) all compiler-related files have moved to ARM\ARMCC\ directory.

SytechDesigns, 2016-07-17:
Thanks Chris, yes I did install the legacy Support pack. I have been thru all the MDK release notes and it looks like they changed the compiler after V4.60 - so any version above that has the 'new' compiler and the libs under arm\armcc - I can now modify my mf4.3 mdktargets file to support all the latter MDK versions.

closed 2016-07-17 by SytechDesigns

cw2, 2016-07-17:
Well, I am not Chris Walker, his GitHub username is cwalker-secretlabs. I just happened to use the same initials when registering on Netduino forums back in 2010 and had to add an extra character to meet the minimum length. I have been using this nick for my embedded endeavors since then...

SytechDesigns, 2016-07-18:
Sorry CW2 , my mistake! Sounds like I'm not the first to make rhe wrong connnection!