NETMF/netmf-interpreter #488: Move CMSIS folder from root

cw2, 2016-08-08:
I would like to suggest changing the location of the CMSIS folder, i.e. moving it from the root to more appropriate place such as `DeviceCode\CMSIS` or perhaps `DeviceCode\Targets\Native\CMSIS` (?). 

And, including vendor-specific device code in its `Device` subdirectory (i.e. remove its entry from .gitignore).

The reason is to unify CMSIS Cortex-Mx Device Peripheral Access Layer Header Files in `Native\**\DeviceCode\` folders (_and by demoting the root CMSIS indicate that NETMF still can be ported to other platforms, even though Cortex is the most popular at the moment_ :).

smaillet-ms, 2016-08-08:
No, let's not move that. In fact the plan for the next release it will be completely removed from the NETMF tree. Instead we'll be using the CMSIS pack approach and CMSIS-5 project files to effectively import CMSIS packages - wherever code is copied for customization it will be done into the equivalent location for what we call a solution at present.

cw2, 2016-08-09:
Ok then.

Is there already a timeframe for the next release?

smaillet-ms, 2016-08-09:
No time frame specifically no, I'm still getting the plan documented based on ideas I've had for many years and the various feedback points from the community. However, it has "begun" in the sense that I just published a clean slate branch with some basic high level goals. The general idea is to document the plan, discuss the plan then execute the plan, review the results and adjust if needed (lather rinse repeat until we have a full feature set). The next major item to document is going to be the coding standards and the contribution process so everyone is clear up front and all code broughtt into the new branch is made to conform from the start. With that in place, once the plan is solid and ready for execution the community can pick up work items and go. The actual time it takes to complete will depend entirely on the level of commitment and contribution from the community.

cw2, 2016-08-09:
That's excellent news!

closed 2016-08-09 by cw2