NETMF/netmf-interpreter #497: vNext copyright?

cw2, 2016-08-13:
As a follow-up to the explanation regarding Microsoft's relations to NETMF as a product in https://github.com/NETMF/netmf-interpreter/issues/491#issuecomment-238939671, I would like to ask for clarification how will Microsoft relate to the vNext project/source code?

At the moment, @smaillet-ms acts as GitHub repository maintainer https://github.com/NETMF/netmf-interpreter/issues/491#issuecomment-239031717 and there is a much larger GitHub team (with invisible members).

For vNext, does this team work on NETMF under the official Microsoft umbrella (i.e. retaining the copyright rights) or is it only a group of individuals contributing to NETMF open-source project? Are there plans to transfer vNext under .NET Foundation?

smaillet-ms, 2016-08-14:
There seems to be some confusion here. There is no "much larger GitHub team (with invisible members)" . The reference to a hidden team was intended to dispel the connection between any real sense of a team by contribution and the GitHub permissions set grouping called a "team". Nothing more. There are a number of people listed in the GitHub Team who are not active contributors to NETMF but would more classically be considered observers their membership in the group is essentially historic at this point. 

The Copyright and lic. of NETMF will remain the same. The Apache 2.0 lic. Governs all of NETMF and any contributions. Changing the lic. would require the agreement of any and all who have contributed to the code base - a virtually impossible task to accomplish even if we had a good reason to do so. 

As previously explained NETMF **IS** under .NET Foundation. (Though not under it's GitHub organization for historical and user impact reasons)  As with any project under Apache, the copyright remains that of the original contributor, however the lic grants fairly liberal rights. For details please read the lic or consult a local legal expert. (I am not a lawyer and am not in a position to provide any legal advise or interpretation on legal matters)

What's the underlying concern here?

cw2, 2016-08-14:
@smaillet-ms Thanks for the information. I just wanted to understand the details.

closed 2016-08-14 by cw2