NETMF/netmf-interpreter #498: MDK uses wrong FPU for Cortex-M7

cw2, 2016-08-18:
```
<FLOATING_POINT_FLAG Condition="'$(PLATFORM_EMULATED_FLOATINGPOINT)'!='true' AND '$(DEVICE_TYPE)'!='Cortex-M4.fp'"> $(SWTC)fpu softvfp </FLOATING_POINT_FLAG>
```
The above line in `Microsoft.Spot.system.mdk.targets` causes software FPU option to be always used for device types other than `Cortex-M4.fp`, e.g. `Cortex-M7.fp.sp`.

I have a PR ready that uses the implicit `--fpu` option for each `--cpu` option (as per [Processors and their implicit Floating-Point Units (FPUs)](http://infocenter.arm.com/help/topic/com.arm.doc.dui0472k/chr1359124234797.html)).

smaillet-ms, 2016-08-22:
Sounds like this could be folded into the other case of FPU handling for the other architectures as a single PR for FP support compiler args.

cw2, 2016-08-24:
@smaillet-ms Could you please explain what do you mean by '_the other case of FPU handling for the other architectures_'?

The intention of this PR is to provide a [simple] hotfix for a bug in MDK targets file that causes invalid use of soft-fpu for device types other than Cortex-M4.fp. I need it for the Cortex-M7 port, but because it affects virtually all device types, I thought it would be a good idea to publish it as a separate fix (so it is immediately available, and other people doesn't need to wait for my Cortex-M7 port, which I'll submit when it is ready).

I am also working on improving FPU support for GCC, but fortunately there are a little different conditions which while not ideal do not really break things. Here I would like to normalize the device type (use same value as for MDK), set the appropriate command line parameters and leave the 'multilib' feature to select the proper libraries. This improvement was supposed to be included in the Cortex-M7 port (*), but it could be folded with the above MDK change as a single PR for FP support compiler args.

(*) My plan is to separate the port (DeviceCode\Targets\Native) and the Nucleo solution into two logical parts, either commits or PRs. I'll create a new issue when things are ready...

smaillet-ms, 2016-08-24:
@cw2 yeah, never mind - I misread the other PR and what it was changing. I need to remember not to read PRs BEFORE the morning caffeine kicks in. 😁

closed 2016-09-05 by cw2